WEC-Sim/WEC-Sim #1181: [WEC-Sim Applications] MOST setting problem

jacky880908, 2023-12-09:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
Hello,
I want to run the application MOST for no-wind and no-wave cases.
So I set the windTurbine(1).control, the wind. mean velocity and the initial rotor speed to 0. Of course the wave height to 0, too.
But somehow the rotor speed started to increase leading the results to no finite, and the simulation stopped. Is there any way I can lock the rotor? Or what would you suggest me to do?
I just want to check the system was stable in an initial setting for the no-wind no-wave case. Thank you.
**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._

**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
[image]
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 2023a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v6.0
[image]
[image]

**Additional context**
_Add any other context about the problem here._

akeeste, 2023-12-13:
Thanks @jacky880908 for bringing this up. I can recreate this crash with the Applications case. 

I expect that the turbine might rotate slightly since the blades are allowed to freely rotate, but am surprised that the case crashes, which is more concerning to me. I'll give another update when I can look into this more this week.

akeeste, 2023-12-13:
Hi @jacky880908

After some initial digging, it seems the issue with the Application case is likely due to the aerodynamic lookup tables (windTurbine.aeroloads.SS). They are not discretized down to 0 m/s, but only to 3.0 m/s. If I run the case with no wave and constant wind speeds from approximately 0-0.75 m/s, my simulations still go unstable, likely for the same reason.

[image]
[image]
[image]

I suggest you rerunning the preprocessing steps in ``mostIO.m`` [here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/MOST/mostData/windTurbine) to account for low velocities. Unfortunately I don't have the expertise to advise on editing these pre-processing steps. @MOREnergyLab will have to advise on going forward with that solution

jacky880908, 2023-12-21:
Hi, @akeeste 
Thank you, it seems better when I replace the rotational joint with a fixed one. The cases didn't crash, and the motion response seems acceptable. 
But now I've found that in the turbine properties of the Blade data. There's a bunch of airfoil data about IEA 15MW turbine.
Do those data come from OpenFAST? 
If so, I want to investigate the NREL 5MW turbine with DeepCwind OC4 Semisubmersible platform. Are there NREL 5MW turbine airfoil data that has already been used in MOST? Cuz, I can't find the data in OpenFAST or on the internet. Thank you.

MassimoSirigu, 2023-12-23:
Hello @jacky880908 ,
If you want to run a static condition with no aerodynamics, I recommend to comment out the aerodynamic lookup tables or put to zero the aerodynamic forces. these issues in MOST will be addressed in the next releases of MOST.

akeeste, 2024-01-10:
Hi @jacky880908 Are there any more clarifications we can give on this issue? I believe we have a few paths forward now:
- replace the hub's rotational joint with a fixed joint
- remove aerodynamic forcing from the library, or update the pre-processing to handle 0 m/s wind speeds
- use the next MOST release that will address this particular issue

@MassimoSirigu will have to comment on whether the NREL 5MW turbine aerodynamic data is readily available. If not, I suggest taking an in-depth look at the pre-processing in the [MOST application case](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/MOST/mostData/windTurbine).

MassimoSirigu, 2024-01-11:
Hello, @akeeste's recommendations seem suitable for addressing your issues. Currently, MOST's present version doesn't offer a direct method to shut down the turbine. One workaround could be to disable the aerodynamic forces. This is due to the current blade element momentum theory (BEMT) implementation, which doesn't always ensure reliable BEMT convergence, especially when creating lookup tables at very low wind speeds or zero rotor speed. We already fixed this issue and a future version of MOST is planned to resolve this by modifying the BEMT resolution approach. Additionally, we're also aware of other minor issues in MOST which will be fixed in the upcoming update.

akeeste, 2024-01-17:
@jacky880908 It has been a few weeks since hearing on this issue, so I will close it. Please reopen the discussion if you have further questions.

closed 2024-01-17 by akeeste