WEC-Sim/WEC-Sim #1182: [WEC-Sim Applications] RM3 MoorDyn Example

dmmavani, 2023-12-12:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**
    First of all, I attempted to run RM3 Moordyn using the lines.txt script that wec-sim had provided, and the simulation was successful.
    But when I was experimenting with it, I changed the lines.txt script. I eliminated a clumped mass that was designated as Free in the 
    "POINTS" section and defined it as a "coupled" body2. Further, body(3) was added in Wecsim which defines the mass properties of 
    the clumped weight based on .txt file and; for inertia, the sphere shaped model was designed in CAD and based on mass properties 
    the inertia was calculated  . After that, the simulation ran, but moorDyn didn't record anything in the lines.out file. 

**To Reproduce**
   1. run wecSim
   2. Go to Mooring folder where lines.txt exists
   3. Open lines.out file
   4. Check for the data

**Expected behavior**
The simulation will begin and MoorDyn will be initialized; however, no data will be shown in lines.out, and 'Nan error detected' will be displayed in lines.log. You will also notice an error in the command window related to MoorDyn's postprocessing, which I guess indicates that the datafile is empty.

**Screenshots**
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10 
 - [MATLAB/Simulink Release]  R2023a
 - [WEC-Sim Release] v6.0

**Additional context**
Here are all the files required for simulation for you to test.
[RM3_Moordyn_1.zip](https://github.com/WEC-Sim/WEC-Sim/files/13652711/RM3_Moordyn_1.zip)

Thank you!

jtgrasb, 2023-12-13:
Hi @dmmavani Thank you for your question. I believe the error you are seeing is because you are trying to use MoorDyn with multiple bodies. Although MoorDyn can handle multiple bodies, WEC-Sim has not yet been set up to be compatible with this feature. 

This compatibility requires WEC-Sim to output the displacements and velocities of each body to MoorDyn and then recieve a force vector from MoorDyn and apply that force to each of the respective bodies. This is a feature that is being currently developed.

RyanDavies19, 2023-12-14:
@dmmavani I will chime in on this to say that if you change Body2's type to 'free' in your lines.txt input file then you should see results correctly output from MD. WECSim is currently only set up to interface with MoorDyn via a single coupled body. 

You will also need to change the output flag you have in your lines.txt file because you are calling for positions of point 10, which doesn't exist in your system
```
-------------------------- OUTPUTS --------------------------------
Point1FX
Point1FY
Point1FZ
Point1MX
Point1MY
Point1MZ
Point1PX
Point1PY
Point1PZ
**Point10PX
Point10PY
Point10PZ**
Point10FX
Point10FY
Point10FZ

```

dmmavani, 2023-12-15:
> Hi @dmmavani Thank you for your question. I believe the error you are seeing is because you are trying to use MoorDyn with multiple bodies. Although MoorDyn can handle multiple bodies, WEC-Sim has not yet been set up to be compatible with this feature.
> 
> This compatibility requires WEC-Sim to output the displacements and velocities of each body to MoorDyn and then recieve a force vector from MoorDyn and apply that force to each of the respective bodies (at each timestep). This is a feature that is being currently developed.

Ohh, I see. Also, Is MoorDyn version 1 compatible with wecsim. I mean can we use version 1 to define lines.txt file?

dmmavani, 2023-12-15:
> @dmmavani I will chime in on this to say that if you change Body2's type to 'free' in your lines.txt input file then you should see results correctly output from MD. WECSim is currently only set up to interface with MoorDyn via a single coupled body.
> 
> You will also need to change the output flag you have in your lines.txt file because you are calling for positions of point 10, which doesn't exist in your system
> 
> ```
> -------------------------- OUTPUTS --------------------------------
> ...
> Point10PX
> Point10PY
> Point10PZ
> Point10FX
> Point10FY
> Point10FZ
> ...
> ```

Thank you so much for this feedback I will take care of this.

RyanDavies19, 2023-12-15:
> Ohh, I see. Also, Is MoorDyn version 1 compatible with wecsim. I mean can we use version 1 to define lines.txt file?

@dmmavani MoorDyn v1 is no longer compatible with the most recent version of WEC-Sim, you have to use v2 and a corresponding v2 input file. You can read more in the MoorDyn [documentation](https://moordyn.readthedocs.io/en/master/inputs.html).

jtgrasb, 2023-12-15:
Thanks @RyanDavies19. To add a bit of clarification, WEC-Sim was recently updated to be compatible with MoorDyn v2; thus, you should use v2 input conventions in the lines.txt file. The feature that has not been added to WEC-Sim yet is the ability to have MoorDyn connections to multiple bodies.

hfchen20, 2023-12-15:
@jtgrasb May I ask when the multiple bodies feature would be available? I have run MoorDyn with multiple bodies: individually moored or interconnected with shared moorings. Let me know if I can help.

> Thanks @RyanDavies19. To add a bit of clarification, WEC-Sim was recently updated to be compatible with MoorDyn v2; thus, you should use v2 input conventions in the lines.txt file. The feature that has not been added to WEC-Sim yet is the ability to have MoorDyn connections to multiple bodies.

[image]

jtgrasb, 2023-12-18:
Hi @hfchen20, I expect the multiple bodies feature to be available within the next few months. The concept itself is not extremely difficult. On the WEC-Sim side, the displacements and velocities from each connection need to be fed into MoorDyn together and the forces will be output together. Then, if the lines.txt file is set up correctly, it will be functional. The main aspect that is taking a bit more time is making sure WEC-Sim's MoorDyn Simulink blocks are still intuitive and plug-and-play. I appreciate you offering your support, I will let you know if I need any help!

closed 2024-01-03 by jtgrasb