WEC-Sim/WEC-Sim #1184: Error when trying to run the WEC-Sim from Simulink GUI [Theory or Implementation]

Dadidal, 2023-12-15:
**Is your question request related to a problem? Please describe.**
When I am trying to run my model from Simulink GUI I get the following error:

```
Variant control 'sv_visualizationOFF' of variant block 'Ship_T14H02h16_WD00_L160W500_flex_cable4_RL/Global Reference Frame/waveVis' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not match the model name.
Component:Simulink | Category:Model error

Variant control 'sv_visualizationON' of variant block 'Ship_T14H02h16_WD00_L160W500_flex_cable4_RL/Global Reference Frame/waveVis' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not match the model name.
Component:Simulink | Category:Model error
```
Also I want to know how to setup the viscous drag matrix in hydrodynamic block mask? I don't see such input field: 
[image]

**Describe the theory or implementation approach that you have a question**
I successfully run my model from wecSimInputFile.m by typing wecSim to command line. But in this case the opened Simulink model closes. I tried to run the model from Simulink as described in [](https://wec-sim.github.io/WEC-Sim/master/developer/library.html#run-from-simulink) but got the abovementioned error. I tried both Parameter Input Methods: "Input file" and "Custom parameters" and got this errors in every case. Also when I am trying to "Load parameters from Input File" I receive another error: 
[image]

**Personal project context**
My end goal is to use the Reinforcement Learning Toolbox to create an RL Agent, which will drive some parts of my model. The training of RL agent includes a lot of simulations, which are carried out by Simulink one by one in automatic mode. In order to do this I want to run my Simulink model without typing wecSim to command line. I guess if I can run WEC-Sim from Simulink I can reach my goal.

**Describe the type of conclusion or resolution you are looking for**
I want to know how to properly run Simulink model with WEC-Sim blocks without typing wecSim to command line. Is it possible to run such model during training the RL agent?

akeeste, 2023-12-18:
Hi @Dadidal 

Thanks for opening an issue and using our template, I'll assist on this issue.

### On you original issue:
The variant control error is likely caused by skipping the initialization of the model when [Running From Simulink](https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#running-from-simulink). After opening the model, but before running in Simulink, you should run ``initializeWecSim`` in the command line. That should eliminate the variant control error. 

### On the "Load parameters from Input File" error:
Double check the path to the wecSimInputFile that is defined in the Global Reference Frame. This seems like a path issue. See what that button does in loadInputFileCallback.m

### On automatic runs in Simulink for RL:
However requiring the command line initialization will likely not meet your goal of automatic runs from Simulink. WEC-Sim's "Run From Simulink" is currently a two-step process. When the feature was developed MATLAB had an order of operations error where a block-level initialization function (``initializeWecSim``) would not necessarily run before another block's variant subsystem initialization. 

### On the drag matrix:
Unforunately maintaining the custom parameter GUI can be pretty intensive, so we have not yet added the drag matrix into the body block GUI. As is, you can turn on the Morison Element and only define 6DOF viscous drag. Otherwise to get the 6DOF drag matrix into the body block's custom parameter GUI, you need to:
- add a mask variable for 6x6 drag in the Rigid Body block
- add the new variable to writeBlocksFromInput.m: https://github.com/WEC-Sim/WEC-Sim/blob/3d4bc12e70cea7c9ede50182eecc8c42565ca37d/source/functions/simulink/mask/writeBlocksFromInput.m#L54-L79
- add the new variable to writeInputFromBlocks.m: https://github.com/WEC-Sim/WEC-Sim/blob/3d4bc12e70cea7c9ede50182eecc8c42565ca37d/source/functions/simulink/mask/writeInputFromBlocks.m#L131-L158

Dadidal, 2023-12-22:
@akeeste thank you for your answer!
I tried to follow your comments but the problem was not solved.
**1. When I tried to run initializeWecSim in the command line I got the following error:**
`Warning: Escaped character '\U' is not valid. See 'doc sprintf' for supported special characters. 
> In writeInputFromBlocks (line 138)
In initializeWecSim (line 77) 
Error: File: wecSimInputFile_simulinkCustomParameters.m Line: 21 Column: 24
Character vector is not terminated properly.

Error in run (line 91)
evalin('caller', strcat(script, ';'));

Error in initializeWecSim (line 78)
        run(inputFile);
 `
This command generated the file wecSimInputFile_simulinkCustomParameters.m with the following contents:

`% WEC-Sim Input File, written with custom Simulink parameters
% 22-Dec-2023 10:18:02

%% Simulation Class
simu = simulationClass(); 
simu.simMechanicsFile = 'Ship_T14H02h16_WD00_L160W500_flex_cable4_RL.slx'; 
simu.rampTime = 5.6; 
simu.endTime = 10; 
simu.solver = 'ode45'; 
simu.dt = 0.01; 
simu.cicEndTime = 10; 

%% Wave Class
waves = waveClass('regular'); 
waves.height = 0.02; 
waves.period = 1.4; 
waves.spread = 0; 

%% Body Class
body(1) = bodyClass('C:\Users\dadid\Documents\MATLAB\WEC-Sim\Projects\AE\hydroData\April\Aqwa_wamit\ANALYSIS.h5'); 
body(1).geometryFile = 'C:body(1).mass = 21.4; 
body(1).inertia = [0.54127 5.5805 6.0524]; 
body(1).nonlinearHydro = 1; 

%% Constraint Class
constraint(1) = constraintClass('constraint1'); 
constraint(1).location = [0 0 0]; 
`
So as I understand there is a problem with parsing the path of geometry file from rigid hydrodynamic body block. I configured this block as follows:
[image]
Do you know how to solve this issue?

**2. I tried to check the path file of Global Reference Frame block pointed to input file but I cannot recognize any problem.**
The path file is provided as follows:
[image]

According to the Figure with error I provided in my previous message there is some problem with variable "type", which is evaluated in "if-else" conditions in loadInputFileCallback.m

**3. Maybe it is stupid question but how can I add mask variable to hydrodynamic block?**
Should I break the link with WEC-Sim library for this purpose?

akeeste, 2024-01-02:
Hi @Dadidal 

Apologies for the delay as our team was largely out of office over the new year.

1. On this point, I agree that the only error in the generated input file is the ``body(1).geometryFile`` line. I am not sure why this is happening, but I can recreate it using the WEC-Sim RM3 example. Based on where the print stops, it has something to do with the colon character breaking a command. I will dig into this and get back to you.

2. Apologies on this error as my original message was not quite correct. This is not an issue with the file path, but rather with the path to blocks within the Simulink model. I'm guessing that you have non-WEC-Sim library blocks at the top-most level of your model, correct? This looks like a bug. ``loadInputFileCallback`` should be skipping blocks that are not apart of the WEC-Sim library, as they don't need to be written to the input file. 
 
As a quick fix, you can update the end of ``loadInputFileCallback.m`` to include the following snippet where type=7 is defined and blocks are not written to the input file in that case.

        elseif any(contains(names,{'cable'}))
            type = 6;
        else
            type = 7;
        end

        if type ~= 7
            writeBlocksFromInput(blockHandle,type,inputFilePath);    
            % write blocks again to account for read only params that are now open
            if type==1 || type==0
                writeBlocksFromInput(blockHandle,type,inputFilePath);
            end
        end

3. Yes in this case you should break the library link on the body blocks. To add a mask parameter, open the body library and unlock the library if locked. On the rigid body block: right click --> Mask --> View Mask (or Ctrl+M). Create a new parameter by selecting the desired type (Edit) on the lefthand menu (Edit, check box, popup, etc). Relocate the mask parameter as needed. See [this page ](https://wec-sim.github.io/WEC-Sim/master/developer/library.html#library-masks) of the developer documentation for more detailed instructions.

I'll work on a bug fix PR for issues 2-3 this week. If you have any feedback on that documentation page, please let me know and I can include that in my PR too.

akeeste, 2024-01-03:
@Dadidal Thank you for bringing these two bugs to our attention. I created PR #1195 to fix problems 1 and 2. See the files changed for an immediate fix. It should be merged into the master and dev branches within ~1 week.

Dadidal, 2024-01-12:
@akeeste Good afternoon! Thank you for providing these detailed instructions.
I tried to edit the `loadInputFileCallback.m` with the code you mentioned but the problem was not solved. The new error appeared instead:
```
Error using initializeWecSim
Invalid Simulink object name: 'RL_3Dship/Global Reference Frame'.
Caused by:
    Error using initializeWecSim
    Unable to find block 'Global Reference Frame'.
```
You are correct that my WEC-Sim blocks in my model are not in the top level. So after this I placed the Global Reference Frame block to the top but the previous error about reading '\U' character still exists. I will wait for the fix, thank you for quick response.

akeeste, 2024-01-15:
Hi @Dadidal Thanks for pointing out these problems. I'll work to make a more robust solution in PR #1195 this week.

akeeste, 2024-01-15:
@Dadidal What MATLAB version and operating system are you using? Also, if possible can you upload your ``wecSimInputFile.m`` and Simulink files for me to test with?

I have fixed the call to the Global Reference Frame Block from ``initializeWecSim``, but I cannot replicate the '\U' character issue with my current fix.

Dadidal, 2024-01-19:
@akeeste I am using Windows 11. Regarding the Simulink files: I am sorry, but this simulation is a part of sponsored colaborative PhD research. Therefore, I am not allowed to freely send this files until any paper with this simulation will be published. Sorry for inconvenience.
I attached WEC-Sim input files here:
`wecSimInputFileGUI.m` - this file I am trying to import using Simulink GUI
`wecSimInputFile_simulinkCustomParameters.m` - this file is generated by WEC-Sim with error. You can see the problem string on 21 line.
[wecSimInputFiles.zip](https://github.com/WEC-Sim/WEC-Sim/files/13992277/wecSimInputFiles.zip)

akeeste, 2024-01-25:
Hello @Dadidal 

Understood, thanks for sending those files. I tested your input files by:
- Built a simple simulink model made of 1 body, 1 constraint and the global reference frame. Some blocks were also put into empty subsystems to check that the Simulink paths work okay. 
- Import ``wecSimInputFileGUI.m`` using the global reference frame "Load input file" button. Note that the ``wecSimInputFileGUI.m`` *must contain an initialized simulation class*, but will load without error once "simu = simulationClass();" is added.
- I updated the body h5 and geometry paths to include "C:\Users\akeeste\..." to test the '\U' character issue. When running ``initializeWecSim``, the new input file is written correctly. I was unable to replicate your supplied ``wecSimInputFile_simulinkCustomParameters.m``

All of my fixes are currently in my branch "bugfix_runFromSim" on PR #1195. From my testing, the two issues you identified are now solved and will be merged into the master and dev branches. You can test the PR by manually copying all four files to your local version of WEC-Sim, or using a command like ``git fetch REMOTE pull/1195/head:pr1195``. Let me know if you still have problems with this feature.

closed 2024-02-28 by MShabara