WEC-Sim/WEC-Sim #1185: Wave Class Initialization

ttammine, 2023-12-16:
**Is your question request related to a problem? Please describe.**
I am trying to troubleshoot my model by running hydrostatic and free decay tests. However, I got the following error message:
Error using initializeWecSim
waves(1).direction outside of the range of available hydro data

**Describe the theory or implementation approach that you have a question**
My wecSimInputFile.m code doesn't include a line for waves(1).direction. According to the manual, it is not required.
But based on the error message it seems that wecSim is searching for some data from the *.h5 file. 
I don't know how to verify if the *.h5 file has all the required data included. On the other hand, I don't understand why nowave case needs waves direction?

**My wecSimInputFile.m:**
%% Simulation Data
simu = simulationClass();               
simu.simMechanicsFile = 'Purha.slx';                         
simu.endTime=40;                        
simu.dt = 0.01;                         
simu.cicEndTime = 15;
simu.explorer = 'off';

%% Wave Information  
% Regular Waves  
waves = waveClass('noWave');
waves.period = 7; %[s]

%% Body Data
% Float
body(1) = bodyClass('../hydroData/PurhaBarge.h5');    	
body(1).geometryFile = '../geometry/FullPurhaBarge2.stl';      
body(1).mass = 'equilibrium';                  
body(1).inertia = [432223745 15688381898 15944683342];    
body(1).initial.displacement = [ 0 0 0];    % Initial Displacement [m]

%% PTO and Constraint Parameters
% Floating (6DOF) Joint
constraint(1) = constraintClass('Constraint1'); 
constraint(1).location = [0 0 0]; 

**My Nemoh.cal file:**
--- Environment ------------------------------------------------------------------------------------------------------------------ 
1025.000000				! RHO 			! KG/M**3 	! Fluid specific volume 
9.810000				! G			! M/S**2	! Gravity 
0.000000                 ! DEPTH			! M		! Water depth
0.	0.              ! XEFF YEFF		! M		! Wave measurement point
--- Description of floating bodies -----------------------------------------------------------------------------------------------
1				! Number of bodies
--- Body 1 -----------------------------------------------------------------------------------------------------------------------
PurhaBarge.dat      ! Name of mesh file
938  1872			! Number of points and number of panels 	
6				! Number of degrees of freedom
1 1. 0.	0. 0. 0. 0.		! Surge
1 0. 1.	0. 0. 0. 0.		! Sway
1 0. 0. 1. 0. 0. 0.		! Heave
2 1. 0. 0. 0.000000 0.000000 2.720000		! Roll about a point
2 0. 1. 0. 0.000000 0.000000 2.7200000		! Pitch about a point
2 0. 0. 1. 0.000000 0.000000 2.7200000		! Yaw about a point
6				! Number of resulting generalised forces
1 1. 0.	0. 0. 0. 0.		! Force in x direction
1 0. 1.	0. 0. 0. 0.		! Force in y direction
1 0. 0. 1. 0. 0. 0.		! Force in z direction
2 1. 0. 0. 0.000000 0.000000 2.7200000		! Moment force in x direction about a point
2 0. 1. 0. 0.000000 0.000000 2.7200000		! Moment force in y direction about a point
2 0. 0. 1. 0.000000 0.000000 2.7200000		! Moment force in z direction about a point
0				! Number of lines of additional information 
--- Load cases to be solved -------------------------------------------------------------------------------------------------------
1 100 0.3 1.3  !Freq type 1,2,3=[rad/s,Hz,s], Number of wave frequencies, Min, and Max (rad/s)
8 0.000000 360.000000		! Number of wave directions, Min and Max (degrees)
--- Post processing ---------------------------------------------------------------------------------------------------------------
1	0.1	10.		! IRF 				! IRF calculation (0 for no calculation), time step and duration
0				! Show pressure
0	0.	180.		! Kochin function 		! Number of directions of calculation (0 for no calculations), Min and Max (degrees)
0	50	400.	400.	! Free surface elevation 	! Number of points in x direction (0 for no calcutions) and y direction and dimensions of domain in x and y direction
1					 ! Response Amplitude Operator (RAO), 0 no calculation, 1 calculated
1					 ! output freq type, 1,2,3=[rad/s,Hz,s]	
---QTF---
0         ! QTF flag, 1 is calculated

**Personal project context**
I am modeling a simple pendulum on a barge.

**Describe the type of conclusion or resolution you are looking for**
I am still unclear on how to troubleshoot using hydrostatic and free decay tests. I would like to know what is the reason behind the error message and how to make a correction.

akeeste, 2023-12-18:
Hi @ttammine 

Thanks for opening an issue and using our template, I'll assist on this issue. 

I'm not super familiar with Nemoh output, but it does seem like the .cal file includes the necessary wave directions (I assume that's 8 evenly spaced points from 0-360). WEC-Sim's ``waveClass.direction`` defaults to 0 so that should not be an issue. When you process the Nemoh output with BEMIO, what wave directions end up included in the hydro structure?

ttammine, 2023-12-18:
Hi,

I have used the bemio.m as it is in example RM3. Thus, it looks like:
hydro = struct();

hydro = readNEMOH(hydro,'..//Nemoh');
hydro = radiationIRF(hydro,60,[],[],[],1.9);
hydro = radiationIRFSS(hydro,[],[]);
hydro = excitationIRF(hydro,157,[],[],[],1.9);
writeBEMIOH5(hydro)
plotBEMIO(hydro)

I am not familiar with the code and what are the placeholders. But when I check the .h5 -file with HDFView it seems that there only one 90 degrees wave direction case:
[image]

I will re-run Nemoh and Bemio and double-check.

akeeste, 2023-12-18:
@ttammine good idea. That direction is definitely causing your issue in WEC-Sim. Unfortunately we cannot help users run BEM codes. However if the problem ends up being in ``readNEMOH.m`` or another BEMIO function, please let me know and I can help out. 

Lastly, I just want to comment that the maximum frequency limit in the calls to ``radiationIRF`` and ``excitationIRF`` (frequency=1.9 rad/s) seems very low--though I do not know the case or BEM data you are working with. If the IRF ends up being noisy or inaccurate, check that limit.

ttammine, 2023-12-19:
Hi,

unfortunately, re-running Nemoh and Bemio didn't help. I tried also to replicate the Oswec case which has different wave directions. Nemoh couldn't run it. It gave Fortran compiler error. But I think the original Oswec case is written to the older version of Nemoh. The original Oswec Nemoh.cal is written as:
--- Load cases to be solved -------------------------------------------------------------------------------------------------------
500	0.04	20.00		! Number of wave frequencies, Min, and Max (rad/s)
10	0.	90.		! Number of wave directions, Min and Max (degrees)

The latest Nemoh version couldn't solve this.

But when I modified (added Freq type):
1 500	0.04	20.00		!Freq type 1,2,3=[rad/s,Hz,s], Number of wave frequencies, Min, and Max (rad/s)
10	0.	90.		! Number of wave directions, Min and Max (degrees)

and added:
--- Post processing ----------------------------------------------------------------------------------------------------------
1					 ! output freq type, 1,2,3=[rad/s,Hz,s]	

the Nemoh solves the problem.
However, when running Bemio for this solution the information on wave direction is lost from the .h5 -file and there is only one case: 90 degrees.

Based on this I think that there might be some version issue with Nemoh solver and Bemio. Can you somehow verify?

MShabara, 2023-12-20:
Hi @ttammine 
Upon further investigation, we have identified some version issues with WEC Sim that require attention. Currently, WEC Sim is utilizing the older NEMOH version, and efforts are needed to ensure compatibility with NEMOH v3.0.0.

Should you prefer a quicker resolution, utilizing the older NEMOH version is a viable option. Alternatively, if you wish to proceed with NEMOH v3.0.0, we are willing to collaborate with you on modifying the readNEMOH function. However, please be advised that this may take some time, given our tight schedules.

Kindly inform us of your chosen option, and we will be more than happy to provide the necessary assistance.

ttammine, 2023-12-28:
Hi, unfortunately, I don't have enough knowledge to contribute to readNEMOH function modification. Thus, I will utilize an older Nemoh version for the time being.

ajay-menon-iitkgp, 2024-01-08:
Hi @MShabara. I came across a similar issue when working with Nemoh v3.0.2 and ended up manually editing my Nemoh.cal file before running BEMIO on it. If it's alright with you and this problem is not being currently addressed by the WEC-Sim team, I'd be happy to help resolve this issue with the readNEMOH function with your assistance.

ajay-menon-iitkgp, 2024-01-09:
@MShabara. I just took a look at the latest version of WEC-Sim, and it appears to already have addressed this version issue. Lines [92 - 111](https://github.com/WEC-Sim/WEC-Sim/blob/df50a039e3af5d4a01ca2f2e436fc15400277b4e/source/functions/BEMIO/readNEMOH.m#L92-L110) read the frequency data depending on how many parameters they encounter, and is compatible for v2 and v3 of Nemoh. Using WEC-Sim v6.0, BEMIO correctly reads a sample Nemoh v3.0.2 output without generating any errors.

One thing to mention is that the example files in the `examples/BEMIO/Nemoh` directory are for an older version of Nemoh. Should a user try to run those input files (Nemoh.cal, mesh.cal, ID.txt) in Nemoh v3, it generates an error. Perhaps this could be mentioned in the documentation to clear up any confusion for new users trying out the examples?

Thank you!

ttammine, 2024-01-10:
Hi @ajay-menon-iitkgp, did you check after running Bemio that the wave direction is correct in the .h5 file? And did you try different wave directions? I have an issue that although I have 0 angle in Nemoh.cal file the .h5 file has 90 degrees.