WEC-Sim/WEC-Sim #1188: [Theory or Implementation]Definition of constraint(i).location, constraint(i).initial.displacement and pto(i).location, pto(i).initial.displacement

hachikoi1, 2023-12-19:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I do not understand why I get 'X' result when running 'Y' simulation [...]_

**Describe the theory or implementation approach that you have a question**
Hi developers, 
Merry Christmas!
I open this issue to seek clarification on the definitions of `constraint(i).location`, `constraint(i).initial.displacement` and `pto(i).location, pto(i).initial.displacement`.

Regarding PTO location, the docs https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#modifying-constraints-and-ptos clarifies that the PTO motions and loads are relative to the local coordinate system. But for a multibody system, I am not sure if the origin of local coordinate system refers to the center of gravity of the base or the follower.

-  For example, the COG of platform (base) is (0,0,-14.4), but the COG of the left WEC (follow) is (-51.75,0,0). Thus the `pto(1).location`                 should be (-51.75,0,14.4) or (0,0,0)?

[image]

- When performing free decay tests, apart from `body.centerGravity` and `body(1).initial`,  is it necessary to setup `constraint(1).location`, `constraint(1).initial.displacement`,  `pto(1).location` and `pto(1).initial.displacement` ? 
From https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#decay-tests, both body, constraint, pto and mooring should have correct initial conditions. But in WEC-Sim free decay test case `\WEC-Sim v6.0\applications\Free_Decay\`, I found the setting of constraint, pto and mooring doesn't seem so necessary.

From my point, `constraint(1).location` should always be `body(1).centerGravity` (base) and `constraint(1).initial.displacement` should be `body(1).initial.displacement`. 

- However, I found `constraint(1).location` and `constraint(1).initial.displacement` have no effect on all results. 
- `pto(i).location and  pto(i).initial.displacement` have no effect on the platform and WEC results, but would affect `ptos(i).forceConstraint`  

How do I set up `constraint(i).location`, `constraint(i).initial.displacement` and `pto(i).location`, `pto(i).initial.displacement` correctly?

Below is my WEC-Sim input file :
```
% UMaine
body(1)                         =  bodyClass(['../IEA-15-240-RWT/HydroData/WEC_Inner' num2str(Control.WECInnerRadius,'%.5f') '_Outer' num2str(Control.WECOuterRadius,'%.5f') '_Height' num2str(Control.WECHeight,'%.5f') '_Draft' num2str(Control.WECDraft,'%.5f') '/iea-15-240-rwt.h5']);
body(1).mass                    =  1.7838E+07;
body(1).inertia                 =  [1.2507E+10 1.2507E+10 2.3667E+10];
body(1).geometryFile            =  'UMaine_Trusses.stl';
body(1).initial.axis            =  [0 1 0];
body(1).initial.angle           =  deg2rad(0);
body(1).initial.displacement    =  [0 0 0];

% WECLeft
body(2)                         =  bodyClass(['../IEA-15-240-RWT/HydroData/WEC_Inner' num2str(Control.WECInnerRadius,'%.5f') '_Outer' num2str(Control.WECOuterRadius,'%.5f') '_Height' num2str(Control.WECHeight,'%.5f') '_Draft' num2str(Control.WECDraft,'%.5f') '/iea-15-240-rwt.h5']);
body(2).mass                    =  WECMass;
body(2).inertia                 =  WECInertia;
body(2).geometryFile            =  'WECLeft.stl';
body(2).initial.axis            =  [0 1 0];
body(2).initial.angle           =  deg2rad(0);
body(2).initial.displacement    =  [0 0 0];

% WECLower
body(3)                         =  bodyClass(['../IEA-15-240-RWT/HydroData/WEC_Inner' num2str(Control.WECInnerRadius,'%.5f') '_Outer' num2str(Control.WECOuterRadius,'%.5f') '_Height' num2str(Control.WECHeight,'%.5f') '_Draft' num2str(Control.WECDraft,'%.5f') '/iea-15-240-rwt.h5']);
body(3).mass                    =  WECMass;
body(3).inertia                 =  WECInertia;
body(3).geometryFile            =  'WECLower.stl';
body(3).initial.axis            =  [0 1 0];
body(3).initial.angle           =  deg2rad(0);
body(3).initial.displacement    =  [0 0 0];

% WECUpper
body(4)                         =  bodyClass(['../IEA-15-240-RWT/HydroData/WEC_Inner' num2str(Control.WECInnerRadius,'%.5f') '_Outer' num2str(Control.WECOuterRadius,'%.5f') '_Height' num2str(Control.WECHeight,'%.5f') '_Draft' num2str(Control.WECDraft,'%.5f') '/iea-15-240-rwt.h5']);
body(4).mass                    =  WECMass;
body(4).inertia                 =  WECInertia;
body(4).geometryFile            =  'WECUpper.stl';
body(4).initial.axis            =  [0 1 0];
body(4).initial.angle           =  deg2rad(0);
body(4).initial.displacement    =  [0 0 0];

constraint(1)                   =  constraintClass('Constraint1');
constraint(1).location          =  [0 0 -14.4];
constraint(1).initial.displacement =  body(1).initial.displacement;

% WECLeft Translational PTO
pto(1)                          =  ptoClass('PTO1');
pto(1).location                 =  [-51.75 0 0] - [0 0 -14.4];
pto(1).hardStops                =  struct('upperLimitSpecify',               'on',...               
                                          'upperLimitBound',                 15-6.25, ... % WEC height is 12.5             
                                          'upperLimitStiffness',             1e10, ...                
                                          'upperLimitDamping',               1e10, ...                
                                          'upperLimitTransitionRegionWidth', 1, ...               
                                          'lowerLimitSpecify',               'on',...               
                                          'lowerLimitBound',                 -13+6.25, ... % WEC height is 12.5                        
                                          'lowerLimitStiffness',             1e10, ...              
                                          'lowerLimitDamping',               1e10, ...               
                                          'lowerLimitTransitionRegionWidth', 1);

% WECLower Translational PTO
pto(2)                          =  ptoClass('PTO2');
pto(2).location                 =  [51.75*cos(pi/3) -51.75*sin(pi/3) 0] - [0 0 -14.4]; 
pto(2).hardStops                =  pto(1).hardStops;

% WECUpper Translational PTO
pto(3)                          =  ptoClass('PTO3');
pto(3).location                 =  [51.75*cos(pi/3) 51.75*sin(pi/3) 0] - [0 0 -14.4]; 
pto(3).hardStops                =  pto(1).hardStops;

for i = 1:length(pto)
    pto(i).initial.displacement =  body(i+1).initial.displacement;
end
```

Best regards,

**Personal project context**
_If this request is related to your project, please describe what you are attempting to do in your project so the WEC-Sim development team knows your end goal._

**Describe the type of conclusion or resolution you are looking for**
_[Ex.] I am still unclear on how an irregular wave train is generated within WEC-Sim and would appreciate being pointed to where in the documentation and code where I can find additional information._

**Additional context**
_Add any other context or screenshots about the feature request here._

hachikoi1, 2023-12-19:
Hi 
I'll add some details. I attempted to simulated with `pto(1).location =  [-51.75 0 0] - [0 0 -14.4];` and `pto(1).location =  [0 0 0];` separately. The `output1.ptos(1).forceConstraint` of the former is 
```
-334.221371160251	23.8128901331434	0	238.954425859301	4174.71741358245	-4.01768798861790e-05
-1059708.62192639	-145.508858595271	0	-1986.54780387140	14085461.7602226	-0.646377699654472
480748.447196352	140.209841256891	0	1919.90135583054	-10549887.7924884	-1.75983779992395
587652.951302230	316.840567523716	0	3954.24361585149	-9418685.21920556	-0.121205685822659
-693319.128337095	87.2464321044346	0	459.172982378572	10913514.4514960	1.45734099271653
-316526.136901519	150.222480178929	0	1213.95750257696	4532354.41364786	-0.650712413388657
437094.797849757	447.816685428875	0	5063.74229699556	-6401070.20795825	-1.43782507851932
237969.467898807	506.579612789377	0	5476.98154257750	-2471040.92782773	0.381519442960927
-26813.5109680308	458.750042097305	0	4539.24111860030	1713963.81785214	1.65542329444920
175883.242360075	541.008746081409	0	5535.22202032245	-1360837.26775893	1.72166876522464
```
, and `output2.ptos(1).forceConstraint` of the latter is 
```
-334.221371160251	23.8128901331434	0	-103.951192057965	-638.070331123979	-1232.31710456705
-1059708.62192639	-145.508858595262	0	108.779759900477	-1174342.39551737	7529.43705460496
480748.447196352	140.209841256898	0	-99.1203582687139	-3627110.15286096	-7257.61912284453
587652.951302230	316.840567523709	0	-608.260556490122	-956482.720453448	-16396.6205750378
-693319.128337095	87.2464321044390	0	-797.175639925409	929719.003441834	-4513.54552041193
-316526.136901519	150.222480178929	0	-949.246211999653	-25621.9577340171	-7774.66406167300
437094.797849756	447.816685428870	0	-1384.81797318032	-106905.118921752	-23175.9512960227
237969.467898807	506.579612789382	0	-1817.76488158960	955719.409915081	-26215.1134424077
-26813.5109680308	458.750042097297	0	-2066.75948760096	1327849.25991250	-23738.6592552407
175883.242360075	541.008746081433	0	-2255.30392324990	1171881.42222615	-27995.4809409489
```
Clearly, the forces between them are the same, with differences in the moments. For the load results at the initial time: 
```
output1.ptos(1).forceConstraint(1,4:6) = output2.ptos(1).forceConstraint(1,4:6) + cross([51.75,0,-14.4],output2.ptos(1).forceConstraint(1,1:3))
```
That is, `pto(1).location =  [0 0 0];` generates a moment on the platform side column, while `pto(1).location =  [-51.75 0 0] - [0 0 -14.4];` generates a larger moment with respect to platform COG. In this context, `pto(1).location =  [0 0 0];` seems to be right.

salhus, 2023-12-19:
Hi @hachikoi1,

Nice to see you here again!

Some general thoughts,

On WEC-Sim, Simscape, Frame, Constraints, and PTOs
----------------------------------------------------------------------------

The Simscape framework should be thought in terms of frames and not as objects. The body blocks are frames that have attached inertia, and the joint blocks (constraint and PTO) are links that describe kinematic relationships between the Base and the Follower Frame. The Constraint and PTO blocks are merely wrappers around Simscape's 'Joint' blocks. the 'Joints' usually have their own conditions, e.g., it will allow motion along the mutual $z-axis$ of the $B$ and $F$ frames.  These 'Joints' can be actuated by a prescribed force or motion. When the 'Joints' don't have a prescribed force or motion and only 'constraint' the relative kinematics of the $B$ and $F$ frames, they are masked as the 'Constraint' block in WEC-Sim; and when the 'Joints' are actuated (using Internal Mechanics), they are masked as 'PTO' blocks. If you open up any of these blocks, and open the 'Joint' block at its core, you will something like this,

[image]

On PTO/Constraint locations, and initial displacements
------------------------------------------------------------

Now to your original question.

If representing a WEC-Sim model as a bond diagram, a single floating body may look something like this,

[image]

So by setting the constraint/PTO locations etc. one sets the point at which the relative kinematic constraints are located, NOT the bodies themselves. For example, a rotation constraint/PTO location will set the location and orientation of the rotational axis. So, the free-decay tests you mentioned (vertical drops), if the translational constraint axes are aligned with the body axes, it won't matter where the joints are constraining the relative kinematics (Constraints) or actuating the system (PTO). However, it will matter, for say, a flap-type device.

[image]

followed by, 

[image]

So, the constraints and PTOs are only defining where and how the $B$ and $F$ interact. While the body is located  at the body(i).location + any initial displacements/rotations, the constraints/PTOs are only restricting/constraining/defining motion at some point (.location/.initial etc.) and manner (3DOF, 6DOF etc.)

More specifically, to your example. The locations for PTO should be thought of as hinges/jigs/fixtures which are rigidly attached to the $B$ and $F$ frames, no matter how far or in what orientation. Since they are rigidly attached, the location will have significance for systems where rotations/torques are important, as the point of application of force changes.

Note, the visualization in the Simscape Mechanics Explorer is merely an aesthetic shell (unless using non-linear hydrostatics). By setting the PTO/Constraint locations, one defines the point of application on the frame being attached to. 

On connection order of Constraints/PTOs
----------------------------------------------
In general terms, WEC-Sim models are not commutative. Although, all kinematics in the WEC-Sim outputs are reported with respect to the Global Frame.

Let me know if this helped, and if not I will be happy to engage more.

Cheers,
Sal

--
Salman Husain, M.Sc., Ph.D.
National Renewable Energy Laboratory (NREL)

hachikoi1, 2023-12-20:
Hi Salman@salhus 
Happy to see you! 
I generally understand your explanation. I post my interpretation below. Please don't hesitate to correct me if I am wrong.

- `constraint.location` is set with respect to the global coordinate origin. `constraint.location + constraint.initial.displacement` determine the final location in the global coordinate system. `constraint.location` and `constraint.initial.displacement` would not affect the results. `constraint.location` is similar to a concept of  rotation center, which has no addition displacement caused by rotation. If `constraint.location = body.centerGravity` and `constraint.initial.displacement = body.initial.displacement`, then `output.constraints.position = output.body.position`; If `constraint.location = [0 0 0]` by default,  `output.constraints.position ~= output.body.position`, and additional coordinate transformations are required to make the equation work.
- Similarly, `pto.location` is set with respect to the global coordinate origin. `pto.location + pto.initial.displacement` determine the final location in the global coordinate system. The base-follower frame inherently creates two local coordinate systems at `pto.location + pto.initial.displacement` in the global coordinate system. One of them attaches on the base and another attaches on the follower. The relative relationship between them can be seen from Simscape Mechanics Explorer
[image]

Again, thank you so much for responding during the holiday season!

Happy Holidays!
Tianyuan

salhus, 2024-01-04:
Hi Tianyuan @hachikoi1,
Thanks for the wishes ! 
Yes, you are right on all points.  
Sorry I was gone on personal time-off.
If this satisfies and confirms your questions, can we close this?

Cheers,
sal

hachikoi1, 2024-01-05:
Hi Salman [salhus](https://github.com/salhus)
This issue can be closed.

Cheers,
Tianyuan

closed 2024-01-05 by salhus