WEC-Sim/WEC-Sim #1189: The BEMIO/NEMOH/OSWEC file can't be used by nemoh solver v3.0.2[BUG]

smallduan, 2023-12-24:
hello 

I am trying to use nemoh to get the hydrodynamic data, when I trying to use the mesh and the Nemoh.cal file in the  examples file BEMIO/NEMOH/OSWEC meet an error as bellow:
[image]
the solver I used is download from nemoh program https://gitlab.com/api/v4/projects/41313230/packages/generic/nemoh/v3.0.2/nemoh-windows-x64-v3.0.2.zip

ajay-menon-iitkgp, 2023-12-27:
Hello @smallduan! This issue arises because the input files for Nemoh v3.0.2 are different from previous versions. You will need to manually update the `Nemoh.cal` file. The error message actually provides a lot of clues about how to fix it.

```
At line 164 of file /builds/lheea/Nemoh/Common/MNemohCal.f90
Fortran runtime error: Bad integer for item 2 in list input
```

Navigating to line 164 of `MNemohCal.f90` indicates that Nemoh v3.0.2 expects 4 parameters for the frequency input in the `Nemoh.cal` file.
```
READ(ufile,*) !--- Load cases to be solved ---------------!
READ(ufile,*) InpNEMOHCAL%waveinput%FreqType,             &
              InpNEMOHCAL%waveinput%NFreq,                &
              InpNEMOHCAL%waveinput%Freq1,                &
              InpNEMOHCAL%waveinput%Freq2
```

However, the example in WEC-Sim's `BEMIO/NEMOH/OSWEC` only uses 3 parameters from a previous version of Nemoh. You need to add `1` for the frequency type so that the new `Nemoh.cal` file looks as follows:
```
--- Load cases to be solved -----------------------------------------------------------
1        500	0.04	20.00	! Frequency type, Number of wave frequencies, Min, and Max (rad/s)
10	0.	90              ! Number of wave directions, Min and Max (degrees)
```

Barring this one change, I haven't come across any other major conflicts when using Nemoh v3.0.2. Until WEC-Sim officially updates its examples, it helps to download the Nemoh source code for this version (available on the same link you shared) and to refer to the [documentation pages](https://lheea.gitlab.io/Nemoh/index.html).

Hope this helped. Cheers!

smallduan, 2023-12-28:
@ajay-menon-iitkgp  Thank you very much!

closed 2024-01-04 by salhus