WEC-Sim/WEC-Sim #1192: Rrror simulating a “Pelamis” WEC

zzq1019, 2023-12-29:
Hello all,
I'm having troubles modeling my device, which likes a "Pelamis" WEC.
Now the problems:
1. The stl geometry file cannot be displayed correctly. I create .stl files from Solidworks model (the unit is m and do not translate STL output data to positive space). In  Mechanics Explorers, the image does not match the actual gravity.
2. In  Mechanics Explorers, the gravity is correct. Therefore, I think the output should be correct. However, the output is inconsistent with AQWA simulation in time-domain.
Could someone explain how to resolve this please? Many thanks in advance
[image]

closed 2024-01-04 by zzq1019

ajay-menon-iitkgp, 2024-01-05:
Hi @zzq1019. Were you able to resolve your issue, and if so, could you please share where the problems occurred? I've faced similar issues when trying to simulate a Pelamis-style WEC.

Thanks and good day!

zzq1019, 2024-01-08:
@ajay-menon-iitkgp  I'm sorry I just noticed your message. We have resolved the issue regarding the display of the .stl geometry file. However, the simulation results are still quite different from Ansys Workbench. We are still trying to solve the problem.

ajay-menon-iitkgp, 2024-01-08:
Thank you for reverting, @zzq1019. Was a difference in WEC-Sim's import units the reason for the improper stl file display (mm instead of m)? If so, did you have to change the geometry mesh to force WEC-Sim to read it in correctly, i.e., `vertices_coordinates * 1000` or was there a simpler alternative?

In the simulations I run, it appears that WEC-Sim reads the mesh in millimeters, even though I know the CAD software exports the data in meters.

I had another question: have you ever compared results between AQWA and any of the open-source BEM tools (Nemoh, Capytaine) for a pelamis-type device?

Cheers,
Ajay