WEC-Sim/WEC-Sim #1193: [WEC-Sim Applications] Query on Simulating Various Wave Energy Converters Using WEC-Sim

sherlocksilver, 2023-12-29:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
The specific WEC-Sim application cases I'm referencing are the RM3 (Point Buoy) and RM5 (Oscillating Surge Wave Energy Converter) as detailed in the WEC-Sim tutorials.

**Describe the bug or issue**
N/A - This is not a bug report but a query regarding the applicability and flexibility of WEC-Sim for different wave energy converter types.

**To Reproduce**
N/A - This is a general inquiry rather than a specific issue reproduction.

**Expected behavior**
N/A - This is a query, not a bug report.

**Screenshots**
N/A

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 2021b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v6.0]

**Additional context**
I am new to WEC-Sim and learning about its capabilities. WEC-Sim tutorials include examples like the RM3 and RM5. My question is: Can WEC-Sim be used to simulate other types of wave energy converters? Specifically, I am interested in knowing if it is possible to simulate devices like Oscillating Water Columns (OWC), Overtopping Devices, Wave Absorbing Devices, Wave Attenuators, Terminators, and Oscillating Point Absorbers.

If the answer is yes, could you please provide or direct me to sample cases or additional tutorials that illustrate how to simulate these types of wave energy converters using WEC-Sim? Any guidance or reference to existing work in this area would be highly valuable.

salhus, 2024-01-04:
Hi @sherlocksilver 
I would recommend checking out  https://github.com/WEC-Sim/WEC-Sim_Applications

cheers,
sal

sherlocksilver, 2024-01-04:
Hello @salhus, 

Thank you for your response. I've gone through the provided link and found information on OWC cases. However, I couldn't locate details on other converters. Could you please share insights or direct me to relevant resources for different converters?

Regards.

salhus, 2024-01-04:
Hi @sherlocksilver ,

I think exploring Simscape might help you come up with more ways to implement different types of devices. I think searching for papers on how other users have used WEC-Sim may give you more intuition. This forum is geared towards resolving issues that users find in the code.
If there are no 'issues' per se, is it okay if I close this issue?

Cheers,
sal

sherlocksilver, 2024-01-04:
Thank you for your assistance, @salhus. I understand if the issue can be closed. As a newcomer, obtaining sample cases or tutorials for different types of converters would greatly aid my learning process. I'm also delving into research papers for more insight. Any guidance or resources would be immensely appreciated. Thank you again for your support.

Regards.

salhus, 2024-01-04:
@sherlocksilver 
Of course. I think if you go through the applications repository, you would be able to use most of the features in WEC-Sim, and in doing so, you will be able to get intuition on how WEC-Sim can be used for different device archetypes.  As to a wider survey of WECs, here is starting point, https://www.sciencedirect.com/science/article/pii/S0960148115001652

Cheers,
sal

closed 2024-01-04 by salhus