WEC-Sim/WEC-Sim #1194: Amount of output power [Theory or Implementation]

HadeelKhalid, 2024-01-02:
hello, I'm wondering how can I calculate the power generated in the OSWEC example? (I'm not talking about the internal power mechanics I'm asking about the electrical output) is there a function or any suggested method to calculate it please clarify? 

thank you

salhus, 2024-01-04:
@HadeelKhalid 
You may need to use blocks from the PTO-Sim library, or set up your own MATLAB function block...
Please follow the recommended format for raising issues in the future.

Cheers,
sal

dforbush2, 2024-01-17:
@HadeelKhalid you may also look at the Applications repository: WEC-Sim_Applications/PTO-Sim/OSWEC
There are two examples of hydraulic PTOs that ultimately end up with electrical power estimates.

salhus, 2024-01-24:
- Closing this issue due to inactivity.

closed 2024-01-24 by salhus