WEC-Sim/WEC-Sim #1196: If the closed-loop systerm can be solved in WEC-Sim

smallduan, 2024-01-04:
Hi 
I'm just getting into WEC-Sim. I would like to use Wec-sim to calculate arrays of offshore PV platforms, I have studied the provided examples and found that each float in these examples has only one connector to the other floats, I would like to ask if it is possible for wec-sim to calculate a Closed-loop structure like the one shown in this picture below.
[image]

salhus, 2024-01-04:
Yes, you can. You might need to use additional Frame blocks from Simscape that will give you additional attachment points because I think the point of application matter here..

Cheers,
sal

--
Salman Husain. MSc, PhD.
National Renewable Energy Laboratory

smallduan, 2024-01-18:
@dforbush2 @salhus Thank you very much, I give my hydrodynamic data bellow, If anyone wants to give it a try they can put up the results and we can discuss them together. 
[Uploading four_barge_data.zip…]()

smallduan, 2024-01-18:
@dforbush2 @salhus Thank you very much, I am trying to calculate

kmruehl, 2024-01-24:
Hi @smallduan  Thank you for your question. We ask users to open GitHub issues using an appropriate template. This is helpful for the WEC-Sim team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

Can you please edit this issue to use the **WEC-Sim applications specific issue or question** template and include more detail about your setup and error message? Thank you.

smallduan, 2024-01-26:
@kmruehl OK I will give the detail data later on

smallduan, 2024-02-01:
@kmruehl I am sorry, I'm currently on winter break and the school's network department has informed us that Remote Desktop is a security risk and banned us from accessing data on the school's servers from home. I will do my best to perfect this example by the end of the month as you requested.

smallduan, 2024-02-26:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
N/A

**Describe the bug or issue**
 a Closed-loop structure like the one shown in this picture below.
[image]
As shown in the figure, the draft of the floats is 4 meters. The center of gravity of each float is 1 meter below the center of gravity of its submerged portion. The exact parameters can be found in the file I provided. The hydrodynamic data were calculated by NEMOH. However, my calculations are not regular and there are some suddenly increasing values.
[image]
The model parameters above were set by me, so they may not be very standardized and lack experiments that can be controlled. I would appreciate it if you could point out any problems in my simulation.

**To Reproduce**
You can simulate it directly using the file I provided

**Expected behavior**
N/A

**Screenshots**
N/A

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10,]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v6.0]

**Additional context**
N/A
[Uploading FourBarge.zip…]()

smallduan, 2024-02-27:
Hi @dforbush2 

Sorry, I re-uploaded the file.

dforbush2, 2024-02-27:
Hi @smallduan 

I'm taking a look at this now. With regard to model structure, this is correct and how I would have done it too. (I can propose an alternative, physically equivalent structure if you're interested, but it requires breaking the library blocks and I'm not sure it will actually enhance performance at all). 

Running with B2B is always challenging because the hydrodynamic matrices use the full 6N x 6N (where N is the number of bodies), so any spikes in any of the coupled degrees of freedom can cause issues. If your 'regular' wave is near any of these spikes in any of these non-sparse matrices, the intermittent large spikes in the time domain solution that you are seeing are not unexpected. Looking at your BEM results even just in surge, heave, and pitch, you definitely have some of these spikes near your frequency of interest. The strangeness in your IRF functions is also the result of these spikes in your BEM data: using these IRF functions will be problematic if you run 'regularCIC', 'irregular', or other waves. Eventually, you will want to revise this BEM data, but a quick fix may be to decrease your solver step size and use a variable step solver, perhaps 'ode23t'.

You may want to clean up your BEM results using badBEMFix_fcn, available on the DEV branch of WEC-Sim in ./source/functions/BEMIO, though it has not been tested on hydrodynamic matrices that are this large (24 x 24 at each frequency and direction, in your case), so please use with caution. BEMRosetta https://github.com/BEMRosetta/BEMRosetta has similar functionality and were cited in development of the recommended function. 

Recall that a BEM solution is wholly inviscid: if an resonance mode exists, particularly between interacting bodies, it will be undamped and its dynamic effect will be hugely overestimated in the BEM solution. 

In short, the reason you are seeing the irregularities in your result is that your BEM solution is non-smooth and has near-singularities at your frequencies of interest. Cleaning up your BEM solutions sufficiently that your IRF functions look more reasonable and then perhaps simulating 'regularCIC' should improve your results and is the recommended solution. In the short term, you may get more regular results by decreasing simulation step size and using a variable step solver.

closed 2024-03-13 by dforbush2