WEC-Sim/WEC-Sim #1201: Changing operational depth of a device

TheWaveMaster, 2024-01-15:
_Hi all,
First of all, I want to thank the team behind wecSim for your amazing work! 
I work for a wave energy developer, and just finalized a wecSim model for our device which we will heavily rely on in the final optimizations before we build and deploy soon!! 

The amount of features that I was able to add because of wecSim are ridiculous, and the model is showing excellent matching against wave tank data!  Big thanks and please keep up the good work!_ 

Now, the theory implementation I want to discuss and would like to hear a few opinions. 
I want to simulate a device which changes its operational submergence depth in sim (gives you a clue to which company I work for). I was wondering if you guys have any idea regarding this. Obviously, it's not possible to change the hydrodatabase in sim (I can't even begin to imagine what will happen to the math of the model lol). Perhaps with the nonlinear hydros, the issue will be entities like the added mass and the radiation damping as basically the device goes from operating submerged to floating at the surface. Wondering what you guys think.

salhus, 2024-01-16:
Hi @TheWaveMaster (cool name lol),

I think what you are describing is the need for the capability to swap out BEM coefficients for large amplitude motions. I would recommend you put in a feature request, and we have a formal format to follow when doing so...  Short answer, it is possible...  but may require some development.  
If you are willing, you could put in the PR !
Basically, you will have to,
0. Load multiple hydro objects and assign them different names, e.g. hydro_0 = load hydro for SWL, hydro_1 = load hydro for 1m draught and so on..
1. Introduce a conditional if statement that checks for the displacements in the main model,
2. Make a mask over the body blocks,
3. In the mask's code initialization, have it write the hydro variable based on the appropriate hydro object from the conditional if statement...

I might try to implement this but can't guarantee a timeline..  Hope that helps ..

Cheers,
sal

Salman Husain, MSc, PhD,
National Renewable Energy Laboratory.

dforbush2, 2024-01-17:
This sounds like something similar to our passive yaw implementation, where we look up the appropriate set of BEM coefficients from the present state of the device. In that case it is its yaw orientation, but we could do something similar with a lookup table based on z-position relative to the free surface.

TheWaveMaster, 2024-01-24:
> Hi @TheWaveMaster (cool name lol),
>
> I think what you are describing is the need for the capability to swap out BEM coefficients for large amplitude motions. I would recommend you put in a feature [...]

Hi @salhus,

Thank you for your reply!
Yes that is exactly what I am looking for. This will open the door for loads of stuff, like simulating worst case scenarios during deployment, if for example the PTO fails while installing the device. 

I see your point, and that might work! I will definitely put in the request and will try something from my end, I have to have a deeper look at marking the body with a mask. 
I am concerned about the transition though, especially in large sea-states.

TheWaveMaster, 2024-01-24:
> This sounds like something similar to our passive yaw implementation, where we look up the appropriate set of BEM coefficients from the present state of the device. In that case it is its yaw orientation, but we could do something similar with a lookup table based on z-position relative to the free surface.
> 
> But I am a little unclear what exactly is being asked here: non-linear hydro will resolve the pressure and non-linear Froude-Krylov on the body relative to the free surface, but the linear portion of the calculation is still based on the BEM coefficients, taken for the singular orientation of the body. My understanding of the request is you would like to also adjust the linear portion of this calculation.
> 
> Can you please provide some more details on this request?

Sorry if I wasn't clear! 
Let me give you a physical understanding of the system a little bit. Our device is a submerged device with its mass smaller than the mass of water it displaces, hence, it always wants to go back to the surface, this way we can maintain the tension in the PTO tethers (it does have more advantages concerning power production). I have attached a picture of the device to better understand the concept. 
I have managed to build a model of CETO (our device) in wecSim, along with 3 PTOs, with MPC control in each of them, and secondary mooring as well. One of the scenarios that I want to simulate for example is the PTOs disconnecting and the device going from being submerged to floating and being kept in place by the secondary mooring. Something like the passive yaw implementation is exactkly what I meant, but all the hydros, and their coefficients shift between the two positions. 
[image]

salhus, 2024-01-24:
Hi @TheWaveMaster,

Thanks for explaining the problem statement, and I was guessing that it was the CETO device. However, as @kmruehl suggested, it would be easier from a project management perspective to include this as a feature request.

dforbush2, 2024-02-26:
@TheWaveMaster FYI this is happening, but represents a significant development effort to do this robustly w/in the WEC-Sim code structure. We will keep you posted of progress.

TheWaveMaster, 2024-02-27:
Awesome thank you!