WEC-Sim/WEC-Sim #1205: Degenerate mass distribution on follower side of prismatic joint [Theory or Implementation]

KevinWyckoff1, 2024-01-22:
Hello,

**Is your question request related to a problem? Please describe.**
I am getting the following error in my simulation:
Error using wecSim (line 45)
['f2b_6W_a/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 0.0.  Stopping simulation.  There may be
a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening the error
tolerances)
Error in AutoRun (line 106)
       wecSim - Show complete stack trace
Caused by:
    Error using wecSim (line 45)
    ['f2b_6W_a/Translational PTO3/Translational PCC (Prismatic Joint)']: 'f2b_6W_a/Translational PTO3/Translational PCC (Prismatic
    Joint)' has a degenerate mass distribution on its follower side. - Show complete stack trace 
Also, I did not get this error when I simulated at a period of 3s, but it began to appear when I changed the period of waves to 4s.

**Describe the theory or implementation approach that you have a question**
I do not think I am doing anything incorrect in terms of setting up the simulation, but am unsure.

**Personal project context**
I am attempting to simulate an array of four two body heaving WECs in regular arrays. Here is my simulink model:
[image]
I have already tried changing the time step and the damping of the PTOs to no avail. I have read some of the previous issues related to this particular error, but did not find any of them helpful to me in particular, though I could easily be missing something.

**Describe the type of conclusion or resolution you are looking for**
I would like to be able to run this simulation. 

**Additional context**
I apologize if this isn't the correct template. I can reupload if someone thinks it should be in a different section. Any other help or suggestions to try would be much appreciated. Thank you.

kmruehl, 2024-01-22:
Hi @KevinWyckoff1 thanks for posting your issue and for using an issue template. The templates helps us better understand the issue and provide the necessary background information. The issue you're posting about is a known issue what happens whenever multiple constraints/ptos are linked together. We're working on a more robust long-term solution, but we can help brainstorm some solutions for you if you upload your WEC-Sim model as a ZIP file here. 

Tagging @salhus and @dforbush2 for awareness.

dforbush2, 2024-01-22:
Hi @KevinWyckoff1 

I think a *zip file would be really helpful here... the connections do look potentially problematic between the seabed reference and the lower body. Are you using body to body interactions? If so, please also share details of your BEM run.

KevinWyckoff1, 2024-01-22:
Thank you for your quick response @dforbush2 @kmruehl . 

I am using ANSYS AQWA to do the BEM. The AQWA model features the four two body WECs all together. Because of file limits I'm just putting the wecsim input file and the slx file in this zip folder, but if having the ASCII hydrodatabase from AQWA would be helpful I can upload that separately. Let me know if there is anything else I can clarify about the setup.
Thanks

[KevinWyckoff_WECsim_GithubIssue.zip](https://github.com/WEC-Sim/WEC-Sim/files/14014649/KevinWyckoff_WECsim_GithubIssue.zip)

dforbush2, 2024-01-22:
Can you please provide the *ah1 and *lis output files from your AQWA run as well? I can use these to build the *h5 locally and this will let me run and better debug your model. 

@kmruehl correct me if I am wrong (I haven't run this personally for some time) but you may also consider if you should be using simu.b2b = 1, that is, turning body-to-body interactions on. If you do not, each body radiation (damping and added mass) constituents will be defined by the 6 x 6 matrix on the block-diagonal of the 6N x 6N BEM data, where N is the number of bodies. This will _exclude_ any contributions to each bodies sum of forces due to hydrodynamic forces arising from the motions of other bodies (these are the off-diagonal 6 x 6 blocks). **While you're debugging your model it is worth leaving this out in any case, since it is likely to slow your simulation a lot** but since you're working with an array of devices it may be physically relevant. 

Something you might try in the interim is to separate the sea-bed reference frame into 4 separate blocks, one connecting to each body. Because these 1) are not rigidly connected and 2) this reference frame contains the solver configuration this should be a physically equivalent but significantly simpler problem "as seen by the solver". This might fix your problem, but as I mentioned I can't confirm this without particular h5 data.

KevinWyckoff1, 2024-01-23:
I'm attaching the AH1 and LIS files. I also tried to implement your suggestion of having different sea-bed reference blocks for each WEC, but I wasn't certain how to do that effectively since it seems like you can't just have four of the global reference frame blocks without having errors. 
[KevinWyckoff_AH1_LIS.zip](https://github.com/WEC-Sim/WEC-Sim/files/14026761/KevinWyckoff_AH1_LIS.zip)

kmruehl, 2024-01-24:
@KevinWyckoff1 thanks for attached your WEC-Sim model and AQWA data. We'll work on debugging this issue and get back to you with some proposed solutions. @dforbush2 yes, I agree that for debugging purpose the body to body interactions should be off.

dforbush2, 2024-02-22:
I admit i'm stumped.
I've tried: 
1) adding point masses on the base side of the common-frame constraints
2) modifying the reference frame to have 4 connecting frames aligning with the four constraint positions w/ and w/o modifying their relative positions in the input file.
3) modifying the relative positions of the constraint/PTO (this is incase the H5 was off and wasn't placing the bodies where they should be.)
4) a bunch of solver types and sim step sizes. 

I ALWAYS get the degenerate mass error on PTO3 follower side. Which is frankly unusual: usually changing solver types/step sizes and reconfiguring as above will at least move the error message to a different offending block if it doesnt solve it outright.

dforbush2, 2024-02-22:
To use a technical term, this is bananas. Delete two of the WECs and it works fine...indicating that structurally your assembly (as well as my attempted variations) creates a well-posed problem. Perhaps the issue enters into the cross-coupling added mass terms on the off diagonals involving the 3rd or 4th WECs? I am surprised the error would emerge in this way if that were the source.

dforbush2, 2024-02-22:
Confirmed. Its your BEM. Approach 2) runs fine if you try a wave period of say, 6 seconds. A wave period of 4s, as you had in your provided files, also runs fine running regularCIC (which is probably the more robust test anyway). 
I'll update tomorrow with a model that works with your BEM and, if I can, diagnose the issue in your data.

KevinWyckoff1, 2024-02-22:
Thank you for all your work on this. I will try your suggestions.

dforbush2, 2024-02-22:
Hi @KevinWyckoff1 

I've included the attached script I used to look at your BEM quickly. Not the most digestible figures but it shows them all at once and should help you see the problem. Requires readH5ToStruct.m from the wecSim distribution.  I'm going to close this issue for now but feel free to re-open if difficulties continue.
[bemCheck.zip](https://github.com/WEC-Sim/WEC-Sim/files/14377279/bemCheck.zip)

closed 2024-02-22 by dforbush2