WEC-Sim/WEC-Sim #1206: Question about MPC in Controls section

DeminLiLi, 2024-01-22:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
_Identify which project this issue is associated with._
problem path: WEC-Sim_Applications-master\Controls\MPC
**Describe the issue**
_A clear and concise description of the issue being encountered and any necessary context._
Hi Sir,

When I used MPC for control calculations, I found that there was a coeff.mat file in the folder. The use of data in this file is in the makePlantModel.m file, but I do not understand how the data in the coeff.mat file is calculated, so I would like to ask about the meaning and calculation method of the data in the coeff.mat file. Thanks.

Demin

kmruehl, 2024-01-24:
Hi @DeminLiLi Thank you for your question and for using the issue template. This is helpful for the WEC-Sim team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

jtgrasb, 2024-01-24:
Hi @DeminLiLi, 
Thank you for your question. The coeff.mat includes the coefficients for frequency dependence, which are calculated by the plotFreqDep.m file. To read more about the MPC controller, please refer to the corresponding [documentation](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#model-predictive-control-mpc) section and accompanying sources. 

If you have further questions, please provide more information regarding your current approach and issues.

DeminLiLi, 2024-01-26:
Hi @kmruehl  and @jtgrasb，

Many thanks for your help. Due to my negligence, I did not view the plotFreqDep.m file. I have re-run the working condition test based on your information. Currently, everything is fine under the test cases. Once again I would like to express my sincerest gratitude.

Demin

jtgrasb, 2024-01-26:
Great, glad we could help!

closed 2024-01-26 by jtgrasb