WEC-Sim/WEC-Sim #1208: Variant Issues when trying to create a new project

pjpatio, 2024-01-23:
Im basing a new project off of the RM3 model but just using a rigid body instead but the same PTO system as in the tutorials 

I keep getting issues regarding variants and not sure what to do with them

Ive tried restarting and the like but still getting these same issues

Ive attached a zip file of the project for reference
[PTO.zip](https://github.com/WEC-Sim/WEC-Sim/files/14023922/PTO.zip)

Variant control 'sv_b_AnalyticalEfficiency' of variant block '[PTO/PTO-SIM/Hydraulic Motor](matlab:open_and_hilite_hyperlink ('PTO/PTO-SIM/Hydraulic Motor','error'))' must be a condition expression or name of a Simulink.VariantExpression object in the global scope workspace of the model. Ensure that the condition does not match the model name.

Component:Simulink | Category:Model error
Variant control 'sv_b_TabulatedEfficiency' of variant block '[PTO/PTO-SIM/Hydraulic Motor](matlab:open_and_hilite_hyperlink ('PTO/PTO-SIM/Hydraulic Motor','error'))' must be a condition expression or name of a Simulink.VariantExpression object in the global scope workspace of the model. Ensure that the condition does not match the model name.

Variant control 'sv_b_hydroBody' of variant block '[PTO/Rigid Body](matlab:open_and_hilite_hyperlink ('PTO/Rigid Body','error'))' must be a condition expression or name of a Simulink.VariantExpression object in the global scope workspace of the model. Ensure that the condition does not match the model name.

Component:Simulink | Category:Model error
Variant control 'sv_b_nonHydroBody' of variant block '[PTO/Rigid Body](matlab:open_and_hilite_hyperlink ('PTO/Rigid Body','error'))' must be a condition expression or name of a Simulink.VariantExpression object in the global scope workspace of the model. Ensure that the condition does not match the model name.

kmruehl, 2024-01-24:
Hi @pjpatio Thank you for your question. We ask users to open GitHub issues using an appropriate template. This is helpful for the WEC-Sim team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "WEC-Sim applications specific issue or question" template and include more details on your setup and error message, I am happy to help. Please maintain the template structure and the issue type in the title.

Thanks again.

pjpatio, 2024-01-30:
Could i follow up this issue?

jleonqu, 2024-02-06:
Hello @pjpatio 
The PTO-Sim example available [here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim/RM3/RM3_cHydraulic_PTO) uses the rigid body blocks. If you want to use the rigid body blocks, it should work just as in the example.
If you want to use the nonHydro option, you need to edit your wecSimInput.m to create the body object, for this you need to start the definition of the body variables with this line: body(#) = bodyClass(''); In addition o this you need to define different parameters for your nonHydro body such as the center of gravity and the displaced volume. There is an example on how to use the nonHydro body option in the applications repo, [here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Nonhydro_Body)
I hope this can help you to continue with your work.

jleonqu, 2024-03-06:
I am closing this issue, but feel free to open a new one in case you have more questions.

closed 2024-03-06 by jleonqu