WEC-Sim/WEC-Sim #1211: [WEC-Sim Applications] Error in building Simulink model

Anulekha-M, 2024-01-30:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
[Error in building Simulink model](https://github.com/WEC-Sim/WEC-Sim/issues/new?assignees=&labels=&projects=&template=wec-sim-applications-specific-issue-or-question.md&title=%5BWEC-Sim+Applications%5D)

**Describe the bug or issue**
I am trying to build the Simulink model for a WEC. Earlier, it was working fine. But after opening the same model and trying to edit, it shows that 'variable body does not exist'. I have tried to reinstall Wec-Sim again, but this error is still happening.

**To Reproduce**
NA

**Expected behavior**
NA

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 11 Pro]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2022b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v6.0]

**Additional context**
NA

MShabara, 2024-01-31:
Hi @Anulekha-M, 
It seems that when the model was initially built, certain variables were stored in the workspace, but they were lost when Matlab was restarted, leading to the errors.

Could you please check if `body(1)` is correctly defined in the `wecSimInputFile.m` file in your project directory? It's possible that this definition is missing or incorrect, contributing to the issues you're facing.

In order to avoid rebuilding the model from scratch, I highly recommend using the closest example model available in [WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications) as a foundation for your project. This approach will save you time and ensure that your model is based on a well-established and functional template.

Anulekha-M, 2024-02-01:
Hi
So, after getting this error, I remodelled it. But the error still persists. 
Secondly, if I happen to make a new folder and shift all the files related to the simulation in that and try to open the *.slx model, will this be a recurring error? 

[image]
This is a new model I am trying to create, but upon adding the rigid body element, it shows the same error even though it is not associated with any wecSimInputFile.m.
Is this some kind of installation error, because it was working fine a couple of months back?

MShabara, 2024-02-01:
Yes, you can not run a WEC-Sim model without associating it to a `wecSimInputFile.m` file.
The rigid body block has undefined parameters, `wecSimInputFile.m` is used to define them.

To run WECSim, make sure you are in the directory that has both the Simulink model and the `wecSimInputFile.m` 
type `wecSim`+Enter in the matlab command window.

I highly recommend using the closest example model available in [WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications) as a foundation for your project.