WEC-Sim/WEC-Sim #1213: [WEC-Sim Applications] PTO-Sim with fixed input not using WEC-Sim

pjpatio, 2024-01-31:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
N/A

**Describe the bug or issue**
Would it be possible to use PTO-Sim such as the model created in the RM3 tutorial but my input would instead be just a signal or a set of repeating data? What i mean by this is i wont be using the Wave part but only tackling the PTO-Sim portion. I wish to just place an input value into the PTO subsystem and then determine the output power based on that value.

Is this possible? and if so id like some help in establishing this system

I was also wondering about the EMOutput of the Electrical Generator. In order to output the power do i just multiple the torqueEM and the shaftspeed? 

Another clarification would be in the RM3 Hydraulic part the Torque input of the hydraulic piston is in MN am i to assume this is Mega Newtons or is it supposed to be mN for milli newtons?

**To Reproduce**
N/A

**Expected behavior**
I wish to be able to base off an input file of force and time and place that in the hydraulic PTO subsystem and have it output the expected power based on force over time

**Screenshots**
N/A

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release] 2023b
 - [WEC-Sim Release] v6.0

**Additional context**
I just want to try and test the output power for different inputs as it would prove beneficial to my research. Im trying to create a WEC but the concept of the model is still in the works so i just want to see if i could generate this amount of force how much power would it generate for me

jtgrasb, 2024-02-07:
Hi @pjpatio thank you for your questions. I will do my best to answer them all.
1. The PTO-Sim blocks can definitely be used without the other WEC-Sim blocks. For example, the hydraulic PTO requires the Response as an input. The WEC response could be replicated with a sine wave signal and its derivative for the position and velocity, respectively. I would recommend outputting the existing Response variable to workspace to make sure you can match the format (e.g. 6 dof signal with position and velocity components)
2. The product of the torqueEM and shaftspeed will give you the mechanical power in the electric generation. To get the electrical power, you should multiply the current and voltage.
3. Which block are you referring to? The Compressible Fluid Piston does not have a torque input.

pjpatio, 2024-02-08:
For # 1 
What do you mean by outputting the response variable to workspace? apologies im not familiar with this method

For # 2
How do i get the current and the voltage? and by doing so when i get the electrical power i can then say that when i input this amount of force i get this amount of electrical power? 

For # 3 

Apologies I mean the PTO Force when outputted has a value of MN as shown below. This is from the input 

[image]

jtgrasb, 2024-02-08:
1. If you open the `RM3_cHydraulic_PTO.slx` file, the Response is fed into the PTO-Sim subsystem.
[image]
In order to provide a "fake" input, you want to make sure you match the shape of the existing Response signal. I was suggesting using the [To Workspace block](https://www.mathworks.com/help/simulink/slref/toworkspace.html) , but you can also use [Determine Signal Dimensions](https://www.mathworks.com/help/simulink/ug/determining-output-signal-dimensions.html) . Here's how I have created a "fake" input as an example: 
[image]
2. The current and voltage are part of the output of the electrical generator. This can be accessed by looking at the output structure in the MATLAB workspace. For this example, output.ptoSim(6).current and output.ptoSim(6).voltage are the current and voltage, respectively.
3. The plot of the force is created by the `userDefinedFunctions.m` script. The standard outputs are in SI units, so if you open this file, you can see the force is divided by 1e6 to get MN (meganewtons).

closed 2024-03-05 by jtgrasb