WEC-Sim/WEC-Sim #1214: [Theory or Implementation]

timmaflan, 2024-01-31:
**Is your question request related to a problem? Please describe.**
In WECSim am running a simulation with a buoy that does not have a PTO system, it simply is connected to a mooring system using moordyn. Unfortunately when I run the parameters that I am supposed to run, the code does not work, and I get errors. For example, in the lines.out folder, the following  output is printed. In this document you can see that most of the nodes are placed at [nan nan nan] which I assume is an error. MoorDyn v2 log file with output level MSG at 'Mooring/lines.log'
WRN C:\Moordyn\Moordyn\source\MoorDyn2.cpp:2005 readOptionsLine(): Warning: Unrecognized option '='
Generated entities:
	nLineTypes  = 1
	nRodTypes   = 0
	nPoints     = 2
	nBodies     = 1
	nRods       = 0
	nLines      = 1
	nFails      = 0
	nFreeBodies = 0
	nFreeRods   = 0
	nFreePoints  = 0
	nCpldBodies = 1
	nCpldRods   = 0
	nCpldPoints = 0
Time integrator = 2nd order Runge-Kutta
No Waves or Currents, or set externally
Creating mooring system...
Initializing coupled Body 0 in 0, 0, 0...
  - Line1:
    ID: 1
    UnstrLen: 47.34
    N   : 39
    d   : 0.062
    rho : 993.683
    E   : 1.78863e+10
    EI  : 0
    BAin: -0.95
    Can : 1
    Cat : 0
    Cdn : 1
    Cdt : 0
    ww_l: -0.927509
Line 1 damping set to 4.86151e+06 Ns = 14677.3 Pa-s, based on input of -0.95
Initialized Line 1
Finalizing ICs using dynamic relaxation (5X normal drag)
ERR C:\Moordyn\Moordyn\source\Line.cpp:752 getStateDeriv(): NaN detected
Line 1
node positions:
0 : [   0    0 -2.8]
1 : [            0             0 -1.28412e+151]
2 : [           0            0 1.36068e+151]
3 : [ nan  nan -inf]
4 : [nan nan inf]
5 : [ nan  nan -inf]
6 : [nan nan nan]
7 : [ nan  nan -inf]
8 : [nan nan nan]
9 : [ nan  nan -inf]
10 : [nan nan nan]
11 : [ nan  nan -inf]
12 : [nan nan inf]
13 : [ nan  nan -inf]
14 : [nan nan inf]
15 : [            0             0 -2.75016e+151]
16 : [           0            0 6.75861e+150]
17 : [            0             0 -4.45882e+149]
18 : [            0             0 -1.79239e+151]
19 : [nan nan inf]
20 : [ nan  nan -inf]
21 : [nan nan nan]
22 : [ nan  nan -inf]
23 : [nan nan nan]
24 : [nan nan nan]
25 : [nan nan nan]
26 : [nan nan nan]
27 : [nan nan nan]
28 : [nan nan nan]
29 : [nan nan nan]
30 : [nan nan nan]
31 : [nan nan nan]
32 : [nan nan nan]
33 : [nan nan nan]
34 : [ nan  nan -inf]
35 : [nan nan nan]
36 : [ nan  nan -inf]
37 : [nan nan inf]
38 : [            0             0 -2.44089e+151]
39 : [  0   0 -50]
ERR C:\Moordyn\Moordyn\source\MoorDyn2.cpp:328 Init(): t = 0.094 s

**Describe the theory or implementation approach that you have a question**
The Buoy is 2m in diameter and 4m in height with a mass of 7772kg. I have created a very simple moordyn simulation, and have been given specific mooring requirements. I am expected to use a single HMPE cable with EA = 1.29 E8 N, mass = 4kg/m, diameter =0.086m. The buoy's draft is approximately 2m, and the water depth is 50m so I would expect that I should have a tether length of 48m. I have set this model up with other types of single leg mooring lines (e.g. chain) so I know the baseline model works, it is just the input parameters for moordyn (Diameter, mass, EA, and unstretched length) that make the file mess up. 

**Personal project context**
This is a personal project, and I would like to know how best to select mooring line parameters to ensure that the model works. I would also like to know why it isn't working if that is the case. For example, is the EA too stiff? is the diameter too small? etc. 

**Describe the type of conclusion or resolution you are looking for**
Currently there isn't much information available on moordyn/readthedocs on how to select mooring parameters, and how to troubleshoot issues with moordyn. If you could give any guidance here on how to work more cohesively with moordyn that would be great. At the moment, I am simply setting up all parameters, and then trying one parameter at a time to vary the simulation.

timmaflan, 2024-01-31:
here is an example of the lines.txt file being used: 
Mooring line data file for MoorDyn in libmoordyn.dll
---------------------- LINE TYPES -----------------------------------------------------
LineType  Diam      Mass/m        EA          BA/-zeta   EI         Cd     Ca    CdAx     CaAx    
(-)       (m)       (kg/m)        (N)          (N-s/-)  (N-m^2)    (-)     (-)    (-)     (-) 
HMPE    0.086         4        1.29e8         -0.95        0       1     1       0     0
---------------------------- BODIES -----------------------------------------------------
ID   Attachment  X0     Y0    Z0     r0      p0     y0     Mass  CG*   I*      Volume   CdA*   Ca
(#)     (-)      (m)    (m)   (m)   (deg)   (deg)  (deg)   (kg)  (m)  (kg-m^2)  (m^3)   (m^2)  (-)
1       Coupled      0    0    0      0      0       0     0.0     0    0        0        0      0
---------------------- POINTS  -----------------------------------------------------
ID      Attachment    X      Y     Z            Mass   Volume     CdA     CA
(#)     (word/ID)    (m)    (m)   (m)           (kg)   (mˆ3)      (m^2)   (-)
1         Body1       0      0    -2.8             0        0    0        0
2         Fix         0      0    -50              0        0    0        0
---------------------- LINES -----------------------------------------------------
ID     LineType  AttachA  AttachB  UnstrLen     NumSegs   LineOutputs
(#)      (-)       (#)      (#)       (m)         (-)       (-)
1        HMPE       1       2         48        39         tp
---------------------- SOLVER OPTIONS-----------------------------------------
0.001    dtM          - time step to use in mooring integration
0        WaveKin      - wave kinematics flag (0=neglect, the only option currently supported)
3.0e6    kBot         - bottom stiffness
3.0e5    cBot         - bottom damping
50       WtrDpth      - water depth
5.0      CdScaleIC    - factor by which to scale drag coefficients during dynamic relaxation IC gen
0.001    threshIC     - threshold for IC con
200.0    TmaxIC
2        writeLog             Write a log file
0        WriteUnits
FrictionCoefficient = 0.74
-------------------------- OUTPUTS --------------------------------
FairTen1
AnchTen1
--------------------- need this line ------------------

timmaflan, 2024-02-01:
Solved the issue by changing the dtM to a higher number.

closed 2024-02-01 by timmaflan

dav-og, 2024-02-02:
strange - usually need to go lower? 🤔

dav-og, 2024-02-02:
ah ok! Good to know, thanks Timma :)