WEC-Sim/WEC-Sim #1215: [Developer Issue] Lets rename WEC-Sim 'master' branch to 'main'

dav-og, 2024-02-01:
**Project**
WEC-Sim

**Describe the issue**
In alignment with our ongoing efforts to foster diversity, equity, inclusion, and accessibility (DEIA) in our community, I propose that we rename our default branch from 'master' to 'main'. This change is a step towards removing language that may be considered exclusive or insensitive, and it reflects a broader shift in the tech community towards more inclusive terminology.

Several leading organizations and many open-source projects have already embraced this change, recognizing the importance of language in creating an inclusive environment. By renaming our default branch, we not only align ourselves with these best practices but also send a clear message about our values and commitment to DEIA.

For more information see: https://github.com/github/renaming

Cheers,
Dave

kmruehl, 2024-02-09:
@dav-og I agree, this is something I've wanted to do for a while. Making this change will have some downstream impacts that we need to address, so I'd like to make sure the rest of the team is aware prior to making this change. Thanks for the suggestion!

akeeste, 2024-02-28:
I'm reopening this issue because there's a few outstanding things to complete:
- CI tests are not running on the main branch, e.g. #1221 did not have any checks run when it went into main
- all github workflows (3) need to be updated to reference the 'main' branch instead of master

When doing the above, there are some merge conflicts from #1221 that need to be fixed

H0R5E, 2024-03-15:
I don't have write permissions on https://github.com/WEC-Sim/MoorDyn, so I can't update that branch name, unfortunately.

akeeste, 2024-03-20:
@H0R5E Sounds good about the documentation link, we'll keep that in mind for the future. Thank you for your help!

@kmruehl can you update the branch name on WEC-Sim/MoorDyn for completion of this issue?

closed 2024-03-27 by akeeste