WEC-Sim/WEC-Sim #1216: [WEC-Sim Applications]

Wilton3737, 2024-02-05:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._
Hi, I just want to recreate a case study similar to RM3, so that i can simulate different result with different dimension of WEC.
these are my stl file.
[stl folder.zip](https://github.com/WEC-Sim/WEC-Sim/files/14164498/stl.folder.zip)

Using this stl file, then I use Capytaine to produce hydrostatic file
this is the capytaine code to produce capytaine code
[capytaine code for plate and buoy.zip](https://github.com/WEC-Sim/WEC-Sim/files/14164592/capytaine.code.for.plate.and.buoy.zip)

then these are the hydrostatic files produced
[hydrastatic folders.zip](https://github.com/WEC-Sim/WEC-Sim/files/14164659/hydrastatic.folders.zip)

using these results, i want to use WecSim to produce power profile
[test.zip](https://github.com/WEC-Sim/WEC-Sim/files/14164734/test.zip)

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
I have go through every step following the tutorial on the website
[image]

When i tried  to run wecSim it states this error
[image]

**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

Just follow the steps in the tutorials
1. run bemio
2. run wecSimInputFile
3. run wecSim

**Expected behavior**
_A clear and concise description of what you expected to happen._
i expect to produce the power profile like this 
[image]

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screenshot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

jleonqu, 2024-02-07:
Hello @Wilton3737 
The problem is that you are using two different Capytaine result files. You should have just one for the two bodies. There is an example on how to get the Capytaine results for the RM3, which is similar to the system that your are planning to model. The example is available [here](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/CAPYTAINE/rm3)

Wilton3737, 2024-02-10:
Hi @jleonqu,
noted and thank you for pointing out the problem. i will proceed to solve the it.
However, just asking, if you can share me the RM3 capytaine code for me to refer that would be very helpful for my project. 

Thank you,
Regards,
Wong Jit Mun

jleonqu, 2024-02-15:
Hi @Wilton3737 
The Capytaine example for the RM3 is available in this [link](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/CAPYTAINE/rm3). The file that you need to check is "rm3.py". That file uses another script called [call_capytaine.py](https://github.com/WEC-Sim/WEC-Sim/blob/master/examples/BEMIO/CAPYTAINE/call_capytaine.py), which is also available in our WEC-Sim repo.

Wilton3737, 2024-02-16:
Hi @jleonqu 
i have downloaded the folder and try to export rm3.nc file using the Capytaine code that you provide
[image]

However, I only successfully exported hydrostatic and KH files but failed to locate .nc file 
[image]

please guide me to find the  .nc file 
Sorry, I am new to python
thank you

Wilton3737, 2024-02-18:
Hi @jleonqu,
thanks, i tried this code and successfully generated rm3.nc from the original one,,, but there seems to be a little delay waiting for the rm3.nc file to be generate.. it took like 1 or 2 days ,, i couldnot find it at first,, is this problem normal?

However, i proceed with my own generated file in stl format 
this is the float
[image]
this is the spar
[image]

its pretty much the same as the rm3 from the example because i need to test out the power output with different dimensions of rm3 

then this is the problem i facing now 
[image]

I am new to python, and find no clue of how to fix this 
please assist me 
thank you very much

Wilton3737, 2024-02-20:
HI @jleonqu,
I hope you're doing well. I wanted to reach out regarding this issue. I am currently working on a tight deadline, and your input or assistance is crucial to moving forward. I understand you likely have other commitments, but if you could prioritize this task, it would be immensely helpful.
If there's anything I can do to support you or provide additional information to expedite the process, please let me know. Your expertise is invaluable to the success of this project, and we're grateful for your contributions.
Thank you in advance for your attention to this matter. I look forward to hearing from you soon.

jleonqu, 2024-02-20:
Hi @Wilton3737 
If you are studying different dimensions of the RM3 you need to create a new .gdf mesh for the new size that you want to study. I recommend to look at the Capytaine repo and the Capytaine documentation, you need to be familiar with the Capytaine workflow first. These are some links for the Capytaine documentation:
https://github.com/capytaine/capytaine 
https://capytaine.github.io/stable/

Wilton3737, 2024-02-22:
Hi @jleonqu, 
Thanks a lot. I have successfully generated result with different diamensions

Wilton3737, 2024-02-24:
Hi @jleonqu 
BTW, may i know where i can find the list of these parameters meaning?
[image]

Thanks

Wilton3737, 2024-02-25:
Hi @jleonqu 
i was able to generate results with different parameters only until i met this problem 
[image]
how to solve this issue, please guide me 

thank you

closed 2024-02-28 by jleonqu