WEC-Sim/WEC-Sim #1217: Calculation of the nonlinear FK force

guangyaow, 2024-02-05:
I am still unclear on how the nonlinear FK force is calculated for a 3D irregular wave field. 

I can see that in waveElevation.m, the direction for each wave mode (the varaibale in the script is 'direction')  are the x and y coordinates (center(:, 1) and center(:, 2)) considered when calculating the wave elevation for each spatial in a 2D (x-y) space. 

Then the calculated elevation (elv) is used as input of nonFKForce.m. While in the function pDis (within nonFKForce.m), the wave direction and the y coordinate for surface (center(:,2)) are not considered to calculate pressure (only center(:,1) is used in the calculation), as shown below:

function f=pDis(center,elv,AH,w,dw,wDepth,deepWater,t,k,phaseRand,typeNum,rho,g)
% Function to calculate pressure distribution
f = zeros(length(center(:,3)),1);
z=zeros(length(center(:,1)),1);
if typeNum <10
elseif typeNum <20
    f = rho.*g.*AH(1).*cos(k(1).*center(:,1)-w(1)*t);
    if deepWater == 0
        z=(center(:,3)-elv).*wDepth./(wDepth+elv);
        f = f.*(cosh(k(1).*(z+wDepth))./cosh(k(1)*wDepth));
    else
        z=(center(:,3)-elv);
        f = f.*exp(k(1).*z);
    end
elseif typeNum <30
    for i=1:length(AH)
        if deepWater == 0 && wDepth <= 0.5*pi/k(i)
            z=(center(:,3)-elv).*wDepth./(wDepth+elv);
            f_tmp = rho.*g.*sqrt(AH(i)*dw(i)).*cos(k(i).*center(:,1)-w(i)*t-phaseRand(i));
            f = f + f_tmp.*(cosh(k(i).*(z+wDepth))./cosh(k(i).*wDepth));
        else
            z=(center(:,3)-elv);
            f_tmp = rho.*g.*sqrt(AH(i)*dw(i)).*cos(k(i).*center(:,1)-w(i)*t-phaseRand(i));
            f = f + f_tmp.*exp(k(i).*z);
        end
    end
end
f(z>0)=0;
end

May I ask why the pressure is calculated without considering wave direction and y coordinate? Your help and support is much apprecaited.

Best,
Guangyao

dforbush2, 2024-02-07:
Hi @guangyaow 

It is possible that #886 is also relevant here.
It appears to me also that the calculation you have highlighted is in error and that is does not account for waves coming from different directions. It is possible this is related to #1133 .

I have labeled BUG and will investigate further and remediate if necessary.

dforbush2, 2024-02-07:
I will make a change to correct this oversight and follow up with a PR. To be clear, this correction will **only** work for plane waves coming from a user specified direction. If the user is adding multiple wave trains as in #1133, nonlinear hydro will not function properly. This will be addressed in a future PR.

dforbush2, 2024-02-07:
fix is in #1219. Many thanks @guangyaow

dforbush2, 2024-02-21:
This issue has been resolved in #1220
Thanks again @guangyaow

closed 2024-02-21 by dforbush2