WEC-Sim/WEC-Sim #1218: [WEC-Sim Applications] Power calculation in PTO-Sim

fanofdw, 2024-02-07:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim/RM3/RM3_cHydraulic_PTO

**Describe the bug or issue**
Hello, WEC-Sim team. 

I have run the PTO-Sim/RM3/RM3_cHydraulic_PTO example code, and I found that the power obtained by voltage * current of the Electric Generator Equivalent Circuit is greater than the power obtained by torque * shaftspeed of the hydraulic motor.

Question 1: Is it reasonable that the electrical power is bigger than the motor power? The trend of the two power curves seems similar.

Question 2: The power obtained by voltage * current of the Electric Generator Equivalent Circuit is the final power generated by RM3 WEC, right?

Thank you for your time. 

**To Reproduce**
1. Add the code below into userDefinedFunctions.m.
```matlab
figure();
plot(output.ptoSim(6).time,-output.ptoSim(6).voltage.*output.ptoSim(6).current/1000)    % U*I is expected to be negative for generating electricity, and the "-" here is just for a better view in comparison.
set(findall(gcf,'type','axes'),'fontsize',16)
xlabel('Time (s)')
ylabel('Power (kW)')
title('Power')
grid on

figure();
plot(output.ptoSim(5).time,output.ptoSim(5).torque.*output.ptoSim(5).shaftSpeed*2*3.14/60/1000)
set(findall(gcf,'type','axes'),'fontsize',16)
xlabel('Time (s)')
ylabel('Power(kW)')
title('motor Power')
grid on
``` 

2. Modify the simulation duration time:
```matlab
simu.endTime= 2000;  
``` 

3. Run wecSim.
4. Check the output figures.

**Expected behavior**
According to the energy transfer path, the final power should be lower than the motor power.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) R2021a Update 8 (9.10.0.2198249)64-bit (win64)
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) Version: 6.0

**Additional context**
_Add any other context about the problem here._
N/A

dforbush2, 2024-02-07:
Hi @fanofdw 

Thanks for your observations. You have highlighted an issue with this model that becomes apparent with the extended simulation time: you are correct that the hydraulic motor power-at-shaft should always exceed the electrical motor (and should always be exceeded by the mechanical power at the prime mover, the relative motion between the buoy and spar). As a note to other devs, this does not seem to be a stability related issue and has been observed with other simulation step sizes and solver types.

I will work with @jleonqu to provide a resolution.

dforbush2, 2024-02-08:
@jleonqu has found an inconsistent sign convention and will fix this with a PR.

jleonqu, 2024-02-08:
Hello @fanofdw 
Thanks for reporting this issue. I have found two inconsistent signs and I have fixed it in this PR #1221 
The correct equations for the block are written in the PR.

Some aspects to consider:

* The 'Electric Generator Equivalent Circuit' block has $R_{load}$ as an input. The input $R_{load}$ can be positive or negative. If $R_{load}$ is positive, that means that the load is used to extract power from the equivalent circuit, but if $R_{load}$ is negative, that means that the load is used as a source of power for the circuit. For example, I changed the load to be a constant value of 4, and I get the following results for power, with the electric power being less than the mechanical and the hydraulic power, as it should be:
 
[image]

* In the example in the applications repo, available [here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim/RM3/RM3_cHydraulic_PTO), there is a PI controller that is used to adjust the speed in the system. The original settings were $\omega = 1000 rpm$, $P=0.001$, and $I = 0.001$. The controller calculates the value of $R_{load}$ necessary to maintain that speed with the given conditions of pressure, flow, and motor parameters. If you plot the results of that example, that's when you see the weird power results. The controller is calculating a negative value of $R_{load}$ in order to maintain the speed with the given conditions. This is the plot for $R_{load}$ and the power for $\omega = 1000 rpm$, $P=0.001$, and $I = 0.001$. As you can see, the electric power is negative when the value of $R_{load}$ is negative.

[image]

* I have updated the example in the WEC-Sim application repo for $\omega = 2000 rpm$, $P=0.002$, and $I = 0.0005$. The update is here in this PR: [PR #58](https://github.com/WEC-Sim/WEC-Sim_Applications/pull/58). For that case, the value of $R_{load}$ is always positive, which means that the controller is able to maintain the speed of the system around $\omega = 2000 rpm$ without any extra power, just power from the wave input. This is the plot for power and the plot for $R_{load}$ for that case:

[image]

fanofdw, 2024-02-08:
Hello @dforbush2 @jleonqu. Thank you so much for all your help. I have no more questions about this. Feel free to close this issue, please. Thank you again!

jleonqu, 2024-02-08:
For hydraulic power we are using Power = Flow Rate x Delta Pressure, and for mechanical power we are using Power = Torque x Shaft Speed

closed 2024-02-08 by jleonqu