WEC-Sim/WEC-Sim #1222: [MoorDyn] Infinitely Increasing Body Velocity and Position in MoorDyn Output File

yman95, 2024-02-10:
**Describe the bug**
After successfully running MoorDyn v2 without crashing, I'm in the process of analyzing results. WEC-Sim is running as expected, and a case of irregular wave with PM spectrum with seed 1 is simulated. Noticing the drift of my geometry in x, there are probably something wrong with the coupling of the mooring. It doesn't appear that the geometry is correctly moored, and the values on nodes and body are off. It's probable that there are erroes in my implementation, if so, please help me correcting it.

Input file: 
Referencing #1167, POINT2 coordinates is changed to (0,0,-5.45) to fix the point at the bottom of the geometry instead of (0,0,0).
The BEM was calculated with a water depth of 200m, therefore the mooring line length was set to 200m as well. Changing unstretched length of LINE1 from 200 to 210 doesn't seem to stop the constant acceleration in Z of the free body.

Nodes' positions of LINE1 remain stationary. Nodes' velocity and wave velocity at nodes are 0 or near 0. Segments' tension seem high or maxed out for given property. 

**WEC-Sim file(s)**
N/A

**To Reproduce**
Run wecSim and check the Mooring folder for outputs

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
MATLAB command line wecSim from wecSimInputFile.m

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Workstation Information (this section MUST BE completed):**
 - Windows 10
 - [MATLAB R2022b]
 - [WEC-Sim 6.0]
 - MoorDyn v2

**Additional context**
WEC-Sim Folder:
https://drive.google.com/file/d/1J3n5DiVdnSEuNrBM-J_8pXG_2-PrddDY/view?usp=drive_link

Appreciate the helps as always,
Yi

jtgrasb, 2024-02-14:
Hi @yman95, thanks for your question. Unfortunately, I was not able to access the attached google drive folder. Please enable access and I will check out your model.

yman95, 2024-02-16:
@jtgrasb hope it works now.

jtgrasb, 2024-02-20:
Hi @yman95, sorry for the delay, I am able to access now. 

First, to address the issue you are observing the body's constant acceleration in the z direction as an output of MoorDyn (BODY1VZ), this is caused by MoorDyn interpreting the body differently than WEC-Sim. As you can see by the WEC-Sim outputs, the body is not actually accelerating. Because you defined the body as "Free" in the `lines.txt` script, MoorDyn tries to apply its own dynamics to the body, leading to MoorDyn interpreting the body at different locations than WEC-Sim. Instead, the body should be defined as coupled according to MoorDyn's [documentation on Body types](https://moordyn.readthedocs.io/en/latest/structure.html#bodies) to allow WEC-Sim to describe the motion. 

Another issue is that the Point #2's attachment is defined as Fixed. Instead, according to the [point attachment types documentation](https://moordyn.readthedocs.io/en/latest/structure.html#points), the attachment should be set to Body1 so that MoorDyn attaches the mooring line to the body rather than a fixed location.

jtgrasb, 2024-02-28:
@yman95 Does this resolve your issues?

yman95, 2024-02-28:
@jtgrasb Yes, MoorDyn works properly now. Appreciate the help! Sorry, I forgot to close the issue.

closed 2024-02-28 by yman95

jtgrasb, 2024-02-28:
No problem! Glad I could help.