WEC-Sim/WEC-Sim #1223: Linear damping coefficient usage in ship hull free decay test

ttammine, 2024-02-12:
**Is your question request related to a problem? Please describe.**
I am simulating a free decay test for a publically available KRISO Container Ship (KCS) hull. The simulation results in too fast movements and I want to add a linear damping coefficient to slow the motion.  In the literature 3% linear viscous damping is used for a ship hull rolling phenomena. 
However, when I apply the linear damping in the Wec-Sim input file I get exactly the same results as without using it. And if I increase the coefficient to 10 % it is the same result.

**Describe the theory or implementation approach that you have a question**
I would like to get an understanding of how to apply linear damping correctly.
According to the manual user must define:
body(i).linearDamping
However, that command does not work directly as such. In one other issue, I found that the command line needs parentheses:
body(1).linearDamping(6,6) = 0.10; 
But in source code bodyclass.m there are no parentheses:
linearDamping {mustBeNumeric} = zeros(6) % (`6x6 float matrix`) Defines linear damping coefficient matrix. Default = ``zeros(6)``.
The same is said in the manual:
(6x6 float matrix) Defines linear damping coefficient matrix. Default = zeros(6).

Thus, I assume numbers inside the parentheses defines the size of the matrix.
But when I tested to run without the parentheses I got an error message:
_Error in port widths or dimensions. 'Output Port 1' of 'KCS/Rigid Body/Hydrodynamic Body/Additional Linear
    Damping Force Calculation/Velocity' is a one dimensional vector with 6 elements._ 
Thus, a vector is expected to be defined instead of a matrix.
Therefore I tried to use:
body(1).linearDamping(6,6) = [0.0 0.0 0.0 0.10 0.0 0.0];
But that led to an error message:
_Unable to perform assignment because the size of the left side is 1-by-1 and the size of the right side is
1-by-6._

**Personal project context**
_N/A._

**Describe the type of conclusion or resolution you are looking for**
_I am still unclear on how to use linear damping and would appreciate supporting solving how to add linear damping for a ship hull roll phenomena._

**Additional context**
Wec-Sim 6.0
Win 11
Matlab 2023b
Nemoh 3.0.2

I have attached the Wec-Sim and Nemoh files to zip below:
[Zip package Free Decay.zip](https://github.com/WEC-Sim/WEC-Sim/files/14237856/Zip.package.Free.Decay.zip)

salhus, 2024-02-12:
Hi @ttammine,

By defining the (6,6) you are defining the 6,6 (the bottom right element) of the matrix, which is the yaw mode. You probably want to define the (1,1) element i.e., the  surge term. I would also suggest familiarization with MATLAB indexing conventions.  https://www.mathworks.com/help/matlab/math/array-indexing.html

Cheers,
sal

salhus, 2024-02-12:
Hi @ttammine,

You may be mixing up linear damping with viscous damping. The linear damping coefficient simply gets multiplied to the velocity 'in that mode'.

ttammine, 2024-02-13:
Hi,

I think that linear damping is viscous damping. According to the manual:
_A viscous damping force in the form of a linear damping coefficient  can be applied to each body by defining the following body class parameter in the WEC-Sim input file (which has a default value of zero):_

Further in the theory manual:
_Linear damping and quadratic drag forces add flexibility to the definition of viscous forcing
Fv=-CvX ... (formula continues with quadratic terms)
where  is the linear (viscous) damping coefficient,_

For simplicity I have used only the linear part. Thus, defined in the Wec-Sim input file only the linear coefficient.

Because BEM (Nemoh in this case) does not solve viscous damping term it must be given in Wec-Sim input file. As the default value is zero there is no viscous damping unless specified in the input file.
But when I look the velocity and the postion values of the Matlab output matrix of my simulation all values are exactly the same regardless of what value I use as a linear (viscous) damping coefficient.  So it seems that the damping stays at default value zero instead of using the given value in the input file.

ttammine, 2024-02-13:
Hi,

I am sorry but I don't fully follow the meaning of the characteristic here.
The body size is 230 meter long, beam 32.2 m and draft 10.8 m.

I tested also using:
body(1).linearDamping(4,4) = 10;
but I didn't see any difference in the velocity or position output values.

ttammine, 2024-03-04:
Hi @salhus ,

I haven't succeeded in solving why my linear damping is not having any effect on the results. Could you try to simulate it with the files I have uploaded?

salhus, 2024-03-06:
Hi @ttammine,

Apologies for my delayed response. Some thoughts,

I think the linear damping you are using is very low. Since you are using a very low value, it is ineffective on the motion dynamics.
As a thought experiment, consider the mass of the body, now even if the ship was moving at 10 m/s (~18 knots), the damping force will be just 100 N..

Cheers,
sal

ttammine, 2024-03-06:
Hi @salhus ,

Thank you for your support! Based on your reply I noticed that even though I had tried different even large values they were too small.
When I set:
body(1).linearDamping(4,4) = 100000000000;
I start to see the difference.

ttammine, 2024-03-06:
Yes, thanks.

closed 2024-03-06 by salhus