WEC-Sim/WEC-Sim #1224: How to compile a wecsim model into executable(.exe) file

guangyaow, 2024-02-14:
Hello, I need to run the wecsim model as a subroutine in a fortran code. So I am wondering how to compile the wecsim model into a executable(.exe) file, so that it can be run as a single app on a platform without matlab. Thanks.

MShabara, 2024-02-28:
Hi @guangyaow  

Here's the process to compile models that include Simulink into standalone executables, specifically for integrating WECSim into a Fortran environment:

1. MATLAB Compiler GUI: Utilize the MATLAB Compiler GUI instead of the command-line interface. Open MATLAB Compiler by typing mcc in the MATLAB Command Window (install it if required).
2. Add Files: In the MATLAB Compiler window, select "Add Files" to include your Simulink model (.slx) and any necessary MATLAB scripts/functions within WEC-Sim or your in-house functions.
3. Configure Compilation Settings: Configure the compilation settings, including the main file by specifying the WEC-Sim initialization file [initializWecSim.m](https://github.com/WEC-Sim/WEC-Sim/tree/main/source/functions).
4. Build: Click on "Build" to initiate the compilation process. MATLAB Compiler will generate the standalone executable (.exe) file based on your WEC-Sim model and associated MATLAB code.
5. Integration with Fortran: Once compiled, integrate the generated executable into your Fortran codebase. You may need to call the executable from within your Fortran code, passing any required input parameters typically found in the `wecSimInputFile.m`.
7. This process allows you to seamlessly integrate WECSim functionality into your Fortran environment, providing you with a standalone executable that can be executed independently.

Regards,

closed 2024-02-28 by MShabara

kmruehl, 2024-03-22:
@MShabara  I think this would be a great topic to add to our new discussions board. It looks like there is a feature to "convert to discission, so I'm going to give it a try.