WEC-Sim/WEC-Sim #1228: [BUG] Nonlinear Hydro failing tests on master branch

jtgrasb, 2024-02-22:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
The nonlinear hydrodynamics cases are not working and causing tests to fail on WEC-Sim Applications master branch. It seems like PR #1220 may have led to some unintentional errors. 

**WEC-Sim file(s)**
`WECSim_Lib_Body_Elements.slx`
`nonFKForce.m`

**To Reproduce**
_Steps to reproduce the behavior:_
1. Run Nonlinear_Hydro cases on WEC-Sim Applications master

**Expected behavior**
Run without errors

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 -  MATLAB R2023b
 - WEC-Sim v6.0

**Additional context**

dforbush2, 2024-02-22:
FYI a properly-functioning #1220 may change the answers from those that are logged in the unit test for comparison. Physically, the implementation in #1220 is correct. I'll re-test locally and see whats going on.

dforbush2, 2024-02-22:
The issue here is a minor typo (dir wasn't added as an argument to pDis at its function definition). And something more serious where the library linking is broken somehow and it is not recognizing the function waveElevation.

dforbush2, 2024-02-22:
This is resolved by #1229 

HOWEVER I did encounter an unusual error during this debug. addWecSimSource failed to add ./source/functions/simulink/model 
to the path. I have been unable to replicate this error but I suspect it resulted in the original issue with #1220. No idea why this happened but a good general reminder to double-check your path before debugging and making PRs

closed 2024-02-26 by jtgrasb