WEC-Sim/WEC-Sim #1230: [WEC-Sim Applications] Discussion on the optimal PTO parameters

hachikoi1, 2024-02-24:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
Hello, guys. I opened this thread for help on PTO optimization.
The problem originated when I added two other common PTO types (constant, quadratic damping) to the translational PTOs. Thus three PTO types can be implemented via the translational PTO.
For the linear damping: `pto.stiffness ~= 0; pto.damping ~= 0; pto.constant = 0; pto.quadDamping = 0;`
For the quadratic damping: `pto.stiffness = 0; pto.damping = 0; pto.constant = 0; pto.quadDamping ~= 0;`
For the linear damping: `pto.stiffness = 0; pto.damping = 0; pto.constant ~= 0; pto.quadDamping = 0;`
`pto.oscillation = 100*(body.mass+body.addedMass);` is used to avoid numerical oscillations of the velocity around 0 for the constant PTO, see _'Numerical benchmarking study of a selection of wave energy converters'_.
[image]

**The problem currently encountered is that, when a very large PTO parameter is applied (this parameter may generate a PTO force that is much larger than the total force of the floating body), the simulation runs normally, but with unphysical results.**

For example,
The RM3 device: `pto.stiffness = 0; pto.damping = 1e3; pto.constant = 0; pto.quadDamping = 0;`
Total force of the float in the heave direction:
[image]
Total force of the spar in the heave direction:
[image]
PTO force:
[image]

The RM3 device: `pto.stiffness = 0; pto.damping = 1e6; pto.constant = 0; pto.quadDamping = 0;`
Total force of the float in the heave direction:
[image]
Total force of the spar in the heave direction:
[image]
PTO force:
[image]

The RM3 device: `pto.stiffness = 0; pto.damping = 1e8; pto.constant = 0; pto.quadDamping = 0;`
Total force of the float in the heave direction:
[image]
Total force of the spar in the heave direction:
[image]
PTO force:
[image]

**It seems that an increasing linear PTO damping increases total force on the float and spar. The PTO force acts as an active force, not a passive damping force. It's non-physical. Could anyone give me some advice on how to determine a reasonable PTO parameter range?** 

**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2023b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v6.0]

**Additional context**
_Add any other context about the problem here._

dforbush2, 2024-02-26:
Hi @hachikoi1 

I'm a little unclear what model you are talking about. Can you please provide a *zip of your simulink run directory or a more specific description of the model you are trying to run? Or if you could please populate these fields of the template.

To Reproduce
Steps to reproduce the behavior:

    Go to '...'
    Click on '....'
    Scroll down to '....'
    See error

hachikoi1, 2024-02-27:
Dear @dforbush2 
Sorry for my unclear description. I'm trying to add more features to the translational PTO. In addition to the original stiffness and linear damping terms, the translational PTO now contains constant and quadratic damping terms. The modifications in the **\WEC-Sim\source\lib\WEC-Sim\WECSim_Lib_PTOs.slx** are as follow:
[image]

For a linear damping PTO used to simulate a permanent magnet linear generator: `pto.stiffness ~= 0; pto.damping ~= 0; pto.constant = 0; pto.quadDamping = 0;`
For a quadratic damping PTO used to simulate a turbine converter: `pto.stiffness = 0; pto.damping = 0; pto.constant = 0; pto.quadDamping ~= 0;`
For a constant PTO used to simulate a hydraulic generator: `pto.stiffness = 0; pto.damping = 0; pto.constant ~= 0; pto.quadDamping = 0;`

`pto.oscillation = 100*(body.mass+body.addedMass);` is used to avoid numerical oscillations of the velocity around 0 for the constant PTO, see 'Numerical benchmarking study of a selection of wave energy converters'.

The modified lib file is attached [here](https://github.com/WEC-Sim/WEC-Sim/files/14416276/objects.zip).

What I am confused is, for constant and quadratic damping PTOs, the instantaneous or total power always increases monotonically with increasing PTO parameters. That is,  the parameter for maximum output power can not be found.

Do you have any insight on this?

Best regards,

Tianyuan

dforbush2, 2024-03-06:
Ah @hachikoi1 I've got to the bottom of this. If you'll note, adding PTO forces of any magnitude are not affecting your velocity, so it seems like they are not actually being delivered through the PTO as intended. 
Let's look at this screenshot of your library block one level up.
[image]

Note that all of your modifications are in the "InternalMechanics" block in grey here actually only output to the "Response" and the "To Workspace" logging blocks (red paths): nothing from this internal mechanics block is applied between the Body-Follower frame (green path). The only parameters that affect that here are the linear damping and stiffness that are parameters _inside_ the Translational PCC blocks (pto.stiffness, pto.damping). Thats why these linear stiffness/damping will affect dynamics, but the additional terms you added do not. Your logged forces always changed as you varied their parameters but the velocity remained the same: these forces were not actually applied to the bodies. 

The best way to do this probably is not to modify the Translational PTO block but instead add a custom function and use it to define the actuation force input to "Translational PTO Actuation Force". Doing this directly to the Translational PTO block, as you attempted, would require modifying the Translational PCC (Prismatic Joint) block from the multi-body library, which is not impossible but best avoided unless there is a need.

hachikoi1, 2024-03-07:
Dear @dforbush2 

Many thanks for your clarification.

The modification is feasible.

Best regards,

Tianyuan

closed 2024-03-07 by dforbush2