WEC-Sim/WEC-Sim #1232: Difficulties attempting to simulate a rotating mass pto.[Theory or Implementation]

NateHixon73, 2024-02-26:
I am trying to simulate a rotating mass power take off, as pictured below.
[image]
I successfully designed and simulated the hydrodynamic hull, and it performed how I expected for the wave range I expect. The problem comes when adding in the second, non-hydrodynamic body to act as the rotating mass portion of the design. When I add the non hydrodynamic body and attach is as a revolving pto (about the yaw axis), the simulation crashes some time in due to a singularity even with no waves. When watching the visual, the two bodies start at rest but quickly start accelerating up and down and eventually bounce in and out of the water. I have attached the folder I am using to run this. Could this be a bug, or am I connecting the two bodies in an incorrect way? Thanks for any help.
[SPHEREwPTO.zip](https://github.com/WEC-Sim/WEC-Sim/files/14406487/SPHEREwPTO.zip)

dforbush2, 2024-02-26:
Hi @NateHixon73 

Thank you for your *zip file. To help us provide more specific advice, can you please edit your issue to include the fields from our issue template, including Matlab version, WEC-Sim version, operating system, etc.? Also a screenshot of your specific error message.

I ask because these degenerate mass issues can sometimes be version issues.

NateHixon73, 2024-02-27:
Done!

dforbush2, 2024-03-05:
Hi @NateHixon73 

Looking at your model, everything looks about right structurally, this is how I would model things as well. I would recommend that your constraint(2) is replaced with a PTO, since this will afford you more flexibility when you go to impose control on this degree of freedom 

The problem, it appears, is your BEM data. I ran bemio in your provided 'hydro' directory and there are a number of issues

1) Most of your radiation damping coefficients on your float are negative, with one especially large negative spike at 0.5 rad/s in heave that is driving the corresponding impulse response function. Same goes for added mass.
2) You have solved at 19 frequencies from 0.2 rad/s to 2 rad/s. While your discretization (0.1 rad/s) is probably ok, for a body of this size I would recommend you increase this range at least to 10 rad/s. At your highest frequency, you should see your radiation damping coefficients approaching zero and your added mass coefficients approaching a constant.

I'm not a NEMOH expert so the support there I can offer is limited but opening these *dat files:
1) you have modeled 562 nodes as degenerate quadrangles. The resulting maximum discretization of ~140 triangular panels is probably insufficient for this geometry. For example, our ellipsoid example in WEC-Sim Applications has >3000 panels. 
2) In my experience a consistent sign error in added mass or damping may suggest an improper normal vector orientation: please ensure these are pointing outward into the fluid. 
3) There appears to be a contradiction here: I see only 562 nodes in your *dat file but your cal file it seems that more were expected? 

For NEMOH specific support, I recommend posting on their help forums. I would strongly recommend revising your BEM: without this, even if we get you running without error I wouldn't have much faith in the results.

NateHixon73, 2024-03-13:
I managed to fix the BEM data and now have a (semi) working simulation! Thanks for the help and guiding me through the process of getting useful BEM data.

NateHixon73, 2024-03-13:
My main problem was that I was trying to get BEM data on a hallow hull and that led to the normals pointing inwards. I made the hull solid, which solved the inward facing normal problem.

closed 2024-03-13 by dforbush2

smallduan, 2025-09-07:
@NateHixon73 Hi
Hello, I'm trying to use the file you uploaded for simulation, but it reports an error right at the beginning. The error says: "World Frame block (mask) does not have a parameter named 'SimscapeInstrumentationLogging'." Could you share a file that you have successfully run?  

Alternatively, if you have any suggestions on how to fix this issue, I'd appreciate your guidance.