WEC-Sim/WEC-Sim #1237: [WEC-Sim Applications] How to produce a normalized power graph

Wilton3737, 2024-03-06:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._
How to produce a normalized power graph
**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
How to produce a normalized power graph
this is the power output graph that i generated using regular wave 
[image]
However, i have seen papers producing normalized graph like this
[image]
do i have to extra processing using other codes/ software or wecSim has its own function to normalise graph
**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

**Expected behavior**
_A clear and concise description of what you expected to happen._
a normalized grraph like this 
[image]

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

dforbush2, 2024-03-06:
Hello @Wilton3737 

To normalize any value, you pick a normalization constant, and divide your value by it, for exampe: normalized power = Power / normalization constant.

There is not one correct way to do it, the decision for a normalization constant is often arbitrary, but frequently is the max power, the rated power, the RMS power, etc. Often normalized power results are presented in publications where the device developer considers their actual power output to be proprietary, but still want to be able to comment on its trends and characteristics.

dforbush2, 2024-03-06:
I would pick some physically meaningful value to be your normalization constant. For instance, if that is a plot of PTO power on a translational PTO, you could let 
`normFactor = max(output.ptos.powerInternalMechanics)`
While we don't have normalized plotting by default, You could add this to your userDefinedFunctions to produce this easily:
`figure; clf;
plot(output.ptos.time,output.ptos.powerInternalMechanics./normFactor)
xlabel('Time(s)')
ylabel('Normalized mech. power')`

for example.

Wilton3737, 2024-03-07:
Hi @dforbush2 
where could i add in this line 
```matlab
normFactor = max(output.ptos.powerInternalMechanics(:,3))
``````
thanks

dforbush2, 2024-03-07:
Please see our documentation on the wecSim workflow http://wec-sim.github.io/WEC-Sim/master/user/workflow.html. You can look at a reference userDefinedFunctions, in, for example
$WEC-Sim/examples/rm3. You can write this code like any other matlab code, and you'll have access to all the simulation workspace variables, most notably, the 'output' structure which you can process and plot as desired.

closed 2024-03-07 by dforbush2