WEC-Sim/WEC-Sim #1238: Inquire about the functions of the oscillating hydrofoil galvanic energy device

sinclairwan, 2024-03-08:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

Dear developers, my research project during my PhD tried to start with the "oscillating hydrofoil tidal flow energy device", because I also used this code developed by you during my master's degree, I feel that it is very practical, and now I want to make sure that this set of code you developed can be simulated for this form of tidal flow energy device? I am very eager for your reply, it will be very important for me, thank you!

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._
N/A
**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
N/A
**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_
N/A
**Expected behavior**
_A clear and concise description of what you expected to happen._
N/A
**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._
N/A
**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._
N/A

sinclairwan, 2024-03-09:
@dforbush2 
Thanks for your reply！It helps me a lot!As you said, WEC-Sim is based on potential flow theory, which has certain limitations for the simulation of oscillating hydrofoils. So can you recommend me which mature open source codebase should I use for research?

dforbush2, 2024-03-09:
I am not an expert in this type of modeling, so I can't provide too much assistance. This is a great resource for smoothed-particle hydrodynamics: https://www.spheric-sph.org/sph-projects-and-codes

best of luck!

closed 2024-03-09 by dforbush2