WEC-Sim/WEC-Sim #1239: [Theory or Implementation] Coupling of  two bodies

Anulekha-M, 2024-03-11:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
I am trying to simulate the effect of adding a submerged body to a floating WEC system.
I am running the simulation only for heavy motion.
For the single body, I made the model as below

[image]

Power curve was as below:
[image]

For the two-body system, I have made the model as shown below.

[image]
Power:
[image]
which is much less than the single body.

The two bodies are connected by a tether (having some stiffness), and then the submerged body is connected to the PTO using another tether.
Theoretically, I know that the power captured for the two-body system should be more than the single body. But when I am running the simulation, I am getting much less power for the two-body system.

**Describe the theory or implementation approach that you have a question**
As the Simulink model developed is a very simplified version, am I missing the way to couple the two bodies, as I still need to incorporate the stiffness of the connecting lines? I have tried using cables to mimic the tethers, but in that case, the power output obtained had a weird graph.
[image]

[image]

Also, I wanted to add the PTO weight and tether pretension, for which it seems the translational actuation PTO block may be useful. However, I saw some issues where the weight of the PTO has been incorporated by defining a rigid body with mass and inertia. Which is the correct approach here?

**Personal project context**
I am trying to compare the output of the two systems by considering the three DOFs and to see how much is the contribution by considering only heave vs considering heave, pitch and surge. I started the simulation with a single DOF system.

**Describe the type of conclusion or resolution you are looking for**
I want to understand how to couple multiple bodies.

**Additional context**
N/A

jtgrasb, 2024-03-11:
Hi @Anulekha-M, thank you for your question. It is a bit difficult to answer without knowing more about the WEC you are trying to model, but I will do my best.

One potential issue is that the PTO in the model with 2 bodies is placed between body(2) and the global reference frame. This means that the PTO is generating power only from the motion of body(2). In most 2-body WECs, the PTO generates power from the relative motion of the 2 bodies. See the [RM3 example](https://wec-sim.github.io/WEC-Sim/dev/user/tutorials.html#rm3-tutorial) case where the translational PTO is located between body(1) and body(2). If you are still encountering issues, I would recommend also looking at the responses of the two bodies within the output variable (e.g., output.body(1).position, output.body(2).position) to see if these are within a range you'd expect.

In terms of the cable implementation, it is again difficult to understand the results from just the power. It also depends on the type of constraints used (for example, translational constraints would allow for displacement which would mean the PTO wouldn't experience any motion), the cable stiffness and damping, etc. Regardless of the configuration, it is important to be sure the PTO is placed in the correct location to measure the relative motion you are trying to capture and analyze the output response in more detail such as the body and pto displacements. 

Lastly, the pto class has a pretension variable, so that can be directly specified in the `wecSimInputFile.m`. In most cases, we assume the PTO has negligible mass. If you'd like to include the mass of the PTO, I'd recommend using the  translational actuation PTO, which allows you to define the force applied by the PTO yourself so you could add an inertia or calculate its effects using a MATLAB script.

Anulekha-M, 2024-03-12:
What I am trying to model is

[image]

So, the PTO is below the second body and I want to understand if this model will indeed produce more power.

jtgrasb, 2024-03-12:
Thank you for providing a diagram. According to the diagram, there are 2 cables. Thus, I would recommend having 2 rigid bodies in your model with a cable connecting them, then another cable connecting the submerged mass to your PTO. The cables should have spherical constraints on either end to allow the ends to rotate. For the [cable class](https://wec-sim.github.io/WEC-Sim/dev/user/api.html#cable-class), you can specify stiffness, pretension, etc. The cable block is not set up to automatically apply taut and slack limits, but it would be possible by [breaking the library link](https://www.mathworks.com/help/simulink/ug/disable-links-to-library-blocks.html) and adding lower and upper limits to the prismatic joint. As shown below, the cable block actually has a PTO block inside. If you open up the PTO block and then the prismatic joint inside it, you can manually add lower and upper limits.
[image]

The PTO inside the cable block is just for implementation purposes and is not set up to generate power like a normal PTO. Thus you need another PTO block. For the PTO, it should be connected in series with the lower cable (and corresponding constraints), and it is set up to detect the relative motion between the base and follower sides. This motion is sensed by the Translational PCC joint inside the PTO block (you can ctrl+u to look inside any locked block). In this case, the base side will be the global reference frame and the follower would be the bottom end of the cable. I expect the general structure of your model to look like: 
[image]

Anulekha-M, 2024-03-14:
I tried the method you suggested.
However, I am now getting the following error:

[image]

If I give a variable time step using ode45, I get
[image]

From the image, it seems the second body is in a free downward motion, which doesn't make sense.
I have built the same Simulink model as in your response.
[image]

I have a few doubts.

1. Why is there a free fall/rigid body motion for the second body? Do I need to put more constraints?
2. When do we give a fixed or variable time step?

jtgrasb, 2024-03-14:
I suspect that the variable step solver is able to avoid the singularity by reducing the time step during that initial period. Reducing the fixed time step would also likely avoid the singularity, but the variable time step solver may offer a speed up by using larger steps when possible. I usually prefer to use fixed step for analysis purposes but either can be used with WEC-Sim. [Here](https://www.mathworks.com/help/simulink/ug/compare-solvers.html) you can read more about the differences. 

I cannot say why the 2nd body is experiencing the constant downward motion without knowing more about the BEM data. When you run BEMIO, I would recommend plotting the results and looking at them closely for the 2nd body. For example, there may be negative terms in the added mass or radiation damping for certain frequencies. You can also check the hydrostatics (mass and hydrostatic stiffness) to make sure they are as expected.

Anulekha-M, 2024-03-15:
I looked at the BEMIO results, and these are the graphs I am getting
[image]
[image]
[image]
[image]
[image]
[image]
From the graphs, I could not see any negative value for added mass or damping.
Maybe I am missing something in the graphs?

jtgrasb, 2024-03-20:
Hi @Anulekha-M Thank you for sharing your BEM plots, they actually look great, so I suspect that is not the main issue. A couple other things to check would be the hydrostatic restoring coefficients and the mass value to make sure it corresponds to the mass of the displaced volume. If you made the mass larger than the displaced volume, the cable will need to have the right parameters to counteract that so that could be another cause. If it is possible, you could share the model as a zip file and I could check it out directly as well.

Anulekha-M, 2024-03-21:
There was a mistake in updating the BEM plots, which I have corrected now. I had taken the submerged body as body 1 in the previous one, which I now have changed to body 2. So, now the lines in the graphs are interchanged. However, the behaviour of the submerged mass remains the same (free fall). I cross-checked the mass, and it is equal to the displaced volume.
I have attached the zip file.
[WEC with cable.zip](https://github.com/WEC-Sim/WEC-Sim/files/14686873/WEC.with.cable.zip)

jtgrasb, 2024-03-27:
Thank you for attaching the zip file. After looking at your model and comparing it to the [cable example](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/main/Cable) on the WEC-Sim Applications repository, I found one major issue with your slx file. Your first cable (cable(1)) is set up with the base connection above the follower connection. Instead, the base connection should be on the bottom (connected to constraint(3)) and the follower connection on the top (connected to constraint(4)). I believe this should solve your issue regarding the free fall of the submerged mass.

Anulekha-M, 2024-04-01:
Thank you for that. I don't know how I missed it.
I am now able to run the simulation using a variable time step.
While using the fixed time step, I am getting the following error.

[image]

Which is similar to issue #448. So, I tried adding inertia to the system between the PTO and constraint and cable and constraint as follows
[image]
It did not work for me.

jtgrasb, 2024-04-03:
I was able to get your model to work with a fixed time step of 0.00001 s. It seems there are large forces at the beginning of the simulation that require a very small timestep to resolve. I would recommend trying to initialize the system closer to its equilibrium positions so that the forces aren't as great. I didn't go through the process to determine the equilibrium, but i did reduce the PTO damping to see if the high PTO force is impacting the timestep needed. I found that if the PTO damping coefficient is reduced, a larger timestep is sufficient, which likely confirms that large initial forces are the "issue" (not necessarily an issue if you are okay with using a small timestep or variable timestep)

jtgrasb, 2024-04-16:
You're welcome! When you say rope, are you referring to the cables? The cable is not exactly the same as the PTO block. Most notably, the cable can go "slack" and provide no force while the PTO provides a force for both negative and positive displacement.  If you want more control over the PTO force applied, you can use the Translational PTO Actuation Force block which allows you to specify the force as an input. Hope this answers your question.

Anulekha-M, 2024-04-18:
Thank you so much. I got a lot of clarity now.

closed 2024-04-18 by jtgrasb