WEC-Sim/WEC-Sim #1243: Issue with Creating ".gdf" Files for WEC-Sim

samir-miloud, 2024-03-27:
Hello everyone,

I'm having trouble generating ".gdf" (GEOMETRIC DATA FILE) for my project using WEC-Sim. I was hoping someone might be able to offer some advice.
I'm using Gmsh for geometry and mesh creation, but I'm not able to find a solution to convert these files to the required format. Can anyone help me understand how I can convert these files, or provide alternative solutions?
Thank you very much.

Best regards,

samir-miloud, 2024-03-27:
Hello everyone,
I am currently working on a research project using WEC-Sim for wave energy conversion system simulation. However, I am facing difficulties with creating ".gdf" (GEOMETRIC DATA FILE) files on Ubuntu.
I am using Gmsh for geometry and mesh creation, but I haven't found a way to generate ".gdf" files compatible with WEC-Sim. Despite my efforts in searching, I haven't been able to find a suitable solution.

Does anyone have any advice on how to create or convert ".gdf" files using Gmsh or other tools on Ubuntu? Or perhaps you could recommend another CAD software that might better meet my needs?
Any help or suggestion would be greatly appreciated. Thank you in advance for your assistance!

Best regards,

dforbush2, 2024-04-01:
Hi @samir-miloud 

I think we may have resolved this via email, I will summarize here for the community:
1) WEC-Sim does not require GDF files: it requires an H5 file containing the hydrodynamic coefficients. It is WAMIT that requires a GDF mesh. This GDF mesh is used by WAMIT to solve the boundary element method equations for this geometry and produce the *out file that, among others, is then used to create the H5 file via tools included in WEC-Sim/source/functions/BEMIO. 
2) GDF files can be created easily in Rhino, and there are a handful of free tools that can also do so. In particular, MeshMagick https://lheea.github.io/meshmagick/command_line/cli_usage.html#converting-a-mesh-file can create a GDF file from a variety of other mesh formats. ***Note: it is specifically not recommended to convert solidworks stl files to gdf (or any other mesh format) intended for BEM. This can cause a variety of issues ***
3) If you are not planning to use WAMIT specifically, you can generate another mesh format. In particular, it is recommended that users without WAMIT access consider using Capytaine (https://capytaine.github.io/stable/) as their BEM solver, which accepts a variety of mesh formats (as well as GDF).

closed 2024-04-09 by dforbush2