WEC-Sim/WEC-Sim #1246: moordyn coulpe with simulink

HANloading, 2024-04-06:
Dear everyone
I got the module on how to couple with simulink through moordyn's user manual as shown in the picture, but I can't use the callbacks to go and couple with my own model model output in real time, how can I use the code to implement iterating the model displacement and velocity at the next moment based on Flines?
[image]

kmruehl, 2024-04-09:
Thank you for your question. We ask users to open GitHub issues using an appropriate template. This is helpful for the WEC-Sim team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

Can you please edit this issue to use the appropriate issue template and include more detail about your setup and error message? Thank you.

jtgrasb, 2024-04-22:
@HANloading Has this issue been resolved? If so, I will close the issue. If not, please edit the issue to use the appropriate template, and I am happy to help you work with WEC-Sim and MoorDyn.

--Best,
Jeff

closed 2024-05-01 by jtgrasb