WEC-Sim/WEC-Sim #1249: Unable to couple moordyn on Linux OS

Guan-CH, 2024-04-11:
Dear team,
I noticed that a few months ago, you uploaded a libmoordyn.so file in WEC-Sim/MoorDyn, but it seems that it has been deleted since then. 
I attempted to download and run the file on Linux OS, but encountered an error which caused Matlab to crash. 
I would appreciate it if you could provide me with more details about the specific reason for the error in your libmoordyn.so file, as well as any instructions on how to successfully run moordyn on Linux.
Thank you in advance for your assistance.
Best regards

jtgrasb, 2024-04-16:
Thank you for your question. We ask users to open GitHub issues using an appropriate template. This is helpful for the WEC-Sim team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

Can you please edit this issue to use the appropriate issue template and include more detail about your setup and error message? Thank you.

Guan-CH, 2024-04-19:
Okay, I've reedited my issue. Thank you for taking time out of your busy schedule to answer my question！ @jtgrasb

jtgrasb, 2024-04-22:
Thank you for your question and using the template, @Guan-CH. The .so file you downloaded which was added to the WEC-Sim/MoorDyn repository previously was removed because it was causing the tests to fail, Unfortunately, the WEC-Sim team does not have direct access to a linux machine, so we were relying on the MoorDyn tool to generate the .so file. It seems the generated .so file did not work, so we chose to pause support for WEC-Sim+MoorDyn on Linux until we can reliably generate a new .so file that works. We are working on addressing this issue currently but have not resolved it yet.

At present, I recommend using the [compiling instructions](https://moordyn.readthedocs.io/en/latest/compiling.html#) on the MoorDyn Documentation. This gives multiple options including pre-compiled releases, compiling as a simple library, and compiling using cmake. We have been working on using the pre-compiled release to generate the .so file, but have had issues with this so far, so you may need to use one of the other options. I apologize for the inconvenience and we hope to be able to support WEC-Sim+MoorDyn in the future without requiring manual compiling. If you are able to compile it, it would be much appreciated if you shared the .so file so we could add this capability back to WEC-Sim for other users. Otherwise, I expect that we will be able to support this feature in the next few months after resolving the compiling issues.

Guan-CH, 2024-04-23:
Thanks for your careful answers and suggestions, I will try using MoorDyn to see if I can generate the required .so file  in another way.

jtgrasb, 2024-05-13:
HI @Guan-CH, we are still working on the WEC-Sim+MoorDyn updates at the moment and have not explored the Linux compatibility yet. I am converting this issue to a discussion to keep it open. We look forward to hearing any updates on generating the .so file.