WEC-Sim/WEC-Sim #1251: An error occurs when running the RM3 example under the examples path

Fredoom8, 2024-04-14:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
Hello, senior, When I run RM3 under the WEC-Sim\examples\ path, the following error message appears:（The wecSim version is 5.0 and the matlab version is R2021）
*******
Error due to multiple causes.
Caused by:
    错误使用 wecSim (第 45 行)
    Variant control 'sv_b1_AnalyticalEfficiency' of variant block 'RM3/Subsystem/Hydraulic Motor' must be a condition
    expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the
    condition does not match the model name.
    错误使用 wecSim (第 45 行)
    Variant control 'sv_b1_TabulatedEfficiency' of variant block 'RM3/Subsystem/Hydraulic Motor' must be a condition
    expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the
    condition does not match the model name.
*********

**WEC-Sim file(s)**
**
[RM3.zip](https://github.com/WEC-Sim/WEC-Sim/files/14971422/RM3.zip)

**

**To Reproduce**

I have tried two methods, whether I type wecSim from the MATLAB command line window to run it, or type initializeWecSim in the command line window and then run it in Simulink, the following error will appear.

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
I expected that I wanted to replace the BEMIO file obtained from WAMIT with reading it from AQWA, but I failed and always reported this error.

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error maessage from the MATLAB command line._
[image]
**
[image]

**

**Workstation Information (this section MUST BE completed):**
 - OS: [ Windows 10]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [version：5.0]

**Additional context**
_There is another question unrelated to this case. If I create a new geometric model from a 3D software such as solidworks, I would like to know how the geometry file .stl file is converted.

akeeste, 2024-04-18:
Hi @Fredoom8,

Thanks for reaching out. As you closed this issue, I assume this problem has been resolved. My guess is that there was a path issue or the ptoSimClass.m was not being instantiated. Let us know if you have any further questions.

Fredoom8, 2024-04-27:
Sorry Sir/Lady, the issue was not resolved and I closed it thinking there was no reply. I did the following as you guessed:
1. i checked the paths and did not find anything wrong.
2. I don't quite understand what you mean by instantiation, but I followed the procedure in Tutorial: RM3 with PTO-Sim and added the parameter settings of the PTO module in the wecSim input file as follows：

> %PTO new blocks
> %Hydraulic Cylinder
> ptoSim(1) = ptoSimClass('hydraulicCyl');
> ptoSim(1).hydPistonCompressible.xi_piston = 35;
> ptoSim(1).hydPistonCompressible.Ap_A = 0.0378;
> ptoSim(1).hydPistonCompressible.Ap_B = 0.0378;
> ptoSim(1).hydPistonCompressible.bulkModulus = 1.86e9;
> ptoSim(1).hydPistonCompressible.pistonStroke = 70;
> ptoSim(1).hydPistonCompressible.pAi = 2.1333e7;
> ptoSim(1).hydPistonCompressible.pBi = 2.1333e7;
> 
> %Rectifying Check Valve
> ptoSim(2) = ptoSimClass('rectCheckValve');
> ptoSim(2).rectifyingCheckValve.Cd = 0.61;
> ptoSim(2).rectifyingCheckValve.Amax = 0.002;
> ptoSim(2).rectifyingCheckValve.Amin = 1e-8;
> ptoSim(2).rectifyingCheckValve.pMax = 1.5e6;
> ptoSim(2).rectifyingCheckValve.pMin = 0;
> ptoSim(2).rectifyingCheckValve.rho = 850;
> ptoSim(2).rectifyingCheckValve.k1 = 200;
> ptoSim(2).rectifyingCheckValve.k2 = ...
>     atanh((ptoSim(2).rectifyingCheckValve.Amin-(ptoSim(2).rectifyingCheckValve.Amax-ptoSim(2).rectifyingCheckValve.Amin)/2)*...
>     2/(ptoSim(2).rectifyingCheckValve.Amax - ptoSim(2).rectifyingCheckValve.Amin))*...
>     1/(ptoSim(2).rectifyingCheckValve.pMin-(ptoSim(2).rectifyingCheckValve.pMax + ptoSim(2).rectifyingCheckValve.pMin)/2);
> 
> %High Pressure Hydraulic Accumulator
> ptoSim(3) = ptoSimClass('hydraulicAcc');
> ptoSim(3).gasHydAccumulator.vI0 = 8.5;
> ptoSim(3).gasHydAccumulator.pIprecharge = 2784.7*6894.75;
> % FlowRateAccumulator = output.ptoSim(3).FlowRate
> % PressureAccumulator = output.ptoSim(3).Pressure
> 
> %Low Pressure Hydraulic Accumulator
> ptoSim(4) = ptoSimClass('hydraulicAcc');
> ptoSim(4).gasHydAccumulator.vI0 = 8.5;
> ptoSim(4).gasHydAccumulator.pIprecharge = 1392.4*6894.75;
> 
> %Hydraulic Motor
> ptoSim(5) = ptoSimClass('hydraulicMotor');
> ptoSim(5).hydraulicMotor.effModel = 2;
> ptoSim(5).hydraulicMotor.displacement = 120;
> ptoSim(5).hydraulicMotor.effTableShaftSpeed = linspace(0,2500,20);
> ptoSim(5).hydraulicMotor.effTableDeltaP = linspace(0,200*1e5,20);
> ptoSim(5).hydraulicMotor.effTableVolEff = ones(20,20)*0.9;
> ptoSim(5).hydraulicMotor.effTableMechEff = ones(20,20)*0.85;
> 
> %Electric generator
> ptoSim(6) = ptoSimClass('electricGen');
> ptoSim(6).electricGeneratorEC.Ra = 0.8;
> ptoSim(6).electricGeneratorEC.La = 0.8;
> ptoSim(6).electricGeneratorEC.Ke = 0.8;
> ptoSim(6).electricGeneratorEC.Jem = 0.8;
> ptoSim(6).electricGeneratorEC.bShaft = 0.8;

**Result: still the same error as before**

> 
[image]

**My wecsim version is 5.0, matlab version is R2021a, Windows 10 Professional.**
**Looking forward to your early reply**

dforbush2, 2024-06-04:
Hi @Fredoom8 

And sorry for the delay!

The issue does seem to be path related: the model you are showing in your screenshots looks like the completed tutorial found here http://wec-sim.github.io/WEC-Sim/main/user/advanced_features.html#pto-sim

but your attached file is the RM3 model from $/WEC-Sim/tutorials/RM3, which is empty, and the wecSimInputFile in this repository (and the additional file called RM3_wecSimInputFile.m) appear to be the mostly unmodified input files from this directory also. 

Please remember to have all of the following in your CASE directory: 
1) your simulink model, for example 'RM3_with_PTOSim.slx' (I'll use this name to refer to the model you show in your screen shot.)
2) your wecSimInputFile, **called exactly 'wecSimInputFile.m'** modified with the PTO-Sim lines you show above. This should name the simulink model in 1) as:
`simu.simMechanicsFile = 'RM3_with_PTOSim.slx'`
3) In your wecSimInputFile, your hydroData and geometry fields in bodyClass should indicate the location of the h5 and the stl file, respectively, relative to the CASE directory.

While you are in your CASE directory, typing `wecSim` should run the simulation.

The error you are seeing is because it appears you attempted to run a Simulink model that included PTO-Sim blocks from a case directory that did not have a wecSimInputFile that included the PTO-Sim lines in your previous comment. This is what @akeeste means by "instantiation": to use a block in your Simulink file you must provide the necessary parameters for it in your wecSimInputFile.

Hope that helps!

-dforbush2

dforbush2, 2024-06-12:
@Fredoom8 

Did this resolve your issue? Please let us know if you are still having troubles.

closed 2024-06-26 by kmruehl