WEC-Sim/WEC-Sim #1254: Performance of the model at resonant frequency

Aiswariak, 2024-05-09:
Hi,
I am analyzing the performance of a single DOF model at resonant frequency. The model is subjected to a regular wave with wave period which is equal to the natural period of the model. I am expecting a higher heave response as the model moves at resonance conditions. But, I am getting a lower value at this wave period. But if I am increasing the wave period than the natural period, higher responses are obtained and by decreasing the period still lesser response are obtained. All the simulations are performed for a same wave height.

I have attached the folder for your reference. I am using WEC-Sim 4.2 version.
[SWEC_resonance.zip](https://github.com/WEC-Sim/WEC-Sim/files/15260564/SWEC_resonance.zip)

Kindly help. Thanks in advance

jtgrasb, 2024-05-15:
Thank you for your question. We ask users to open GitHub issues using an appropriate template. This is helpful for the WEC-Sim team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

Can you please edit this issue to use the appropriate issue template and include more detail about your setup and how you determined the natural period? Thank you.

Aiswariak, 2024-05-16:
Thank you for the response. I have raised the issue under " Theory or implementation clarification". I do not understand clearly on how to use the appropriate issue template. Kindly help on this regard as well.

I have estimated the heave natural frequency using the theoretical expression as well as by doing a heave decay analysis. I am also getting a maximum RAO value in heave at this frequency from the hydrodynamic diffraction study in ANSYS AQWA.

Thank you.

jtgrasb, 2024-05-17:
Hi @Aiswariak, I got a chance to look at your attached WEC-Sim model. When I remove the pto damping, I see no issues. The free decay shows a natural period of around 0.5 s. This is also matched by the response amplitude across the range of frequencies.
[image]
[image]

With a pto damping of 20, your system is very damped. [Adding damping to the system](https://eng.libretexts.org/Bookshelves/Electrical_Engineering/Signal_Processing_and_Modeling/Introduction_to_Linear_Time-Invariant_Dynamic_Systems_for_Students_of_Engineering_(Hallauer)/10%3A_Second_Order_Systems/10.02%3A_Frequency_Response_of_Damped_Second_Order_Systems) increases the natural period (decreases natural frequency) so it is not surprising that you are getting higher responses for higher periods. If this does not resolve your issue, please show me your method for calculating the natural period. 

A few other tips:
- Your stl file is very large. I believe it was exported in mm rather than m (which is what WEC-Sim reads it in as). You should go back to where you created the stl and make sure you select m when you export. Also, make sure they stl origin is located at the body's cog. The stl will only impact the visualization for a linear model and not the dynamics.
- You should decrease the domain size (simu.domainSize) since your floating body is so small. This will make visualization easier.
- If you want to run any multiple condition runs (MCR) cases, you will have to remove the clear at the beginning of the wecSimInputFile.

Aiswariak, 2024-05-20:
Hi @jtgrasb,
Thank you for the detailed response. This solves my issue. I will take care of the tips as well.

Thanks and regards

closed 2024-05-20 by Aiswariak