WEC-Sim/WEC-Sim #1257: Incorrect Units in h5 Files [Developer Issue]

jniffene, 2024-05-16:
**Project**
BEMIO

**Describe the issue**
I've noticed some of the units listed in the h5 files are incorrect, for instance the added mass at infinite frequency is normalized by the water density but is listed as having units in kg when it should be in m3 and the hydrostatic linear restoring stiffness does not have listed units but is normalized by water density and the acceleration of gravity so it should have units of m2. It would also be good to mention when values are normalized in their descriptions in the h5 files.

MShabara, 2024-07-26:
@jniffene Hi James,

Thank you for raising this issue, I checked and saw the problems in some units needs correction.
Could you please fix them and make a Pull request?

kmruehl, 2024-09-04:
@jniffene thanks for bringing this up, this issue was addressed in your PR https://github.com/WEC-Sim/WEC-Sim/pull/1296 and in https://github.com/WEC-Sim/WEC-Sim/pull/1306

closed 2024-09-04 by kmruehl