WEC-Sim/WEC-Sim #1259: [Developer Issue] MoorDyn version auto updating

RyanDavies19, 2024-05-23:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
MoorDyn

**Describe the issue**
Hi all, this is in relation to pr #1212 

We just released MoorDyn-C 2.3.0 and will need to update the https://github.com/WEC-Sim/MoorDyn library accordingly. Ideally though we should set up github actions to automate this every time there is a new release. I know @jtgrasb has helped deal with compatibility issues in the past, so some form of testing regime would be good. Throwing this up as an issue so we can discuss potential plans for getting this done.

RyanDavies19, 2024-05-23:
Also, let me know if there is a better avenue than an issue to go about this.

jtgrasb, 2024-06-12:
@RyanDavies19 Thank you for making us aware of the new updates. Making our MoorDyn library automatically update is definitely something we are interested in, but also have to be careful if there are any changes that may impact compatibility with WEC-Sim. I am going to move this to a discussion as we work on the automatic updates.