WEC-Sim/WEC-Sim #1260: [WEC-Sim Applications]MOST /IEA15MW FOWT Free Decay

wind2021, 2024-05-30:
[WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/main/MOST) 

**Describe the bug or issue**
I think the WEC-Sim+MOST model may have some problem, as the free decay test for IEA15MW FOWT shows some strange results. I used the model download from the link above and set the initial state of the six degrees of freedom to 0, with no wind and no waves, and carried out the free decay test. The results were surprising, because the surge, heave and pitch did not stabilize near 0, and the other three degrees of freedom were not balanced. Here are the 6 DOFs results.
[image]
[image]
[image]
[image]
[image]
[image]

**To Reproduce**
1. Go to WEC-Sim_Applications/MOST
/mostData/mostIO.m to run it and generate file for MOST;
2. Go to WEC-Sim_Applications/MOST/mostData/hydroData/bemio.m to run it and generate file for MOST;
3. Go to WEC-Sim_Applications/MOST/OFWTmodel.slx to separate the Aerodynamics + Control block;
4. Go to WEC-Sim_Applications/MOST/wecSimInputFile.m to set no wave and no wind, omega0 = 0. And set the body(1).initial.displacement = [0 0 0];
5. Run WEC-Sim.

**Screenshots**
[image]
[wecSimInputFile.txt](https://github.com/WEC-Sim/WEC-Sim/files/15495521/wecSimInputFile.txt)

**Expected behavior**
We should expect a response of six degrees of freedom around zero, fluctuating up and down at very small values, 0.001 or something else.

**Workstation Information (this section MUST BE completed):**
 - [MATLAB/Simulink Release][2022b]
 - [WEC-Sim Release] [v6.0]

**Additional context**
After we change the parameters, according to the report(https://www.nrel.gov/docs/fy20osti/76773.pdf)), in WEC-Sim_Applications-6.0\MOST\hydroData\VolturnUS15MW_nemoh\Mesh\Hydrostatics.dat and WEC-Sim_Applications-6.0\MOST\mostData\windTurbine\turbine_properties\WTproperties.m, the response of six degrees of freedom do come around zero. But the structure becomes very strange. Results are here:
[Hydrostatics.txt](https://github.com/WEC-Sim/WEC-Sim/files/15495666/Hydrostatics.txt)
[WTproperties.txt](https://github.com/WEC-Sim/WEC-Sim/files/15495670/WTproperties.txt)
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Dear friends, can you provide some information? Is this a problem with the model or is there something wrong with my operation?

closed 2024-06-05 by wind2021

akeeste, 2024-06-05:
Hi @wind2021. I see that you closed your issue yesterday. Are you still having problems with this topic or did you find a solution?

kangjianfei, 2025-08-10:
I’m seeing significant sway, heave, and pitch responses in the IEA-15-240-RWT-UMaineSemi model even in still water, no wind conditions, starting from an initial state of 0,0,0,0,0,0 (m, rad).

Could this be inherent to the reference case (e.g., due to imbalance between PtfmVol0 and total mass, RNA CG offset, or mooring pretension not matching the zero pose), or is it likely due to my setup?

I’d like to request reopening this question so I can share more details and investigate further.