WEC-Sim/WEC-Sim #1267: [BUG] Simulation of boat shape 6DOF errors "Derivative of state is not finite"

madewelr, 2024-06-26:
**Describe the bug**
I am simulating a boat shaped body in WEC-Sim in 6DOF. An idea of the body in question can be found here: https://axys.com/ocean-sentinel-nomad-metocean-buoy/
I used WAMIT and Bemio to generate the .h5 file and no errors are thrown in this process. When running the WEC-Sim code, the simulation terminates early between 0.95-1.5sec and displays the below error: 

Error using wecSim
Derivative of state 'nomad.Floating_6DOF.CONSTRAINT_Bushing_Joint.Px.v' in block 'nomad/Floating (6DOF) /CONSTRAINT (Bushing Joint)' at time 1.2000000000000002 is not finite. The simulation will be stopped. There may be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances)

I tried this for JONSWAP waves, regular waves, and no waves. I've changed the step size to try and find one that works. I changed the duration of the simulation. I also changed the simu.solver type. The error occurs in all cases, though with some small variance in how long the simulation runs before erroring.

**WEC-Sim file(s)**
Please see the attached simulation with all files needed to run in WEC-Sim. The error occurs in the 6DOF block in Simulink.
[OS.zip](https://github.com/user-attachments/files/15981046/OS.zip)

**To Reproduce**
1. Extract folder into WEC-Sim.
2. Make sure the Source folder is added to the path.
3. Execute via the wecSim command in the command window in the OS directory.
4. Error will appear in the command window.

I run WEC-Sim from the command window.

**Expected behavior**
The simulation should free decay slightly until it reaches equilibrium as there are no waves in the provided case. Though the issue persists when wave cases are present.

**Screenshots**
1. WEC-Sim window with the error.
[image]

2. Simulink setup with the error.
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 11
 - MATLAB: 2022b
 - WEC-Sim Release: v6.0

**Additional context**
This issue presents very similarly to Bug #1179. As such, I suspect the issue lies in my mesh, but I don't understand what the issue is as WAMIT and BEMIO had no issues processing the mesh.

MShabara, 2024-06-26:
Hi @madewelr,

Thank you for your question. Your model goes unstable pretty rapid, its an indication that the metacentric height is negative from the beginning of the simulation.  

I took a look at the .stl file and hydrodynamic coefficients I noticed that there are some issues

1. The centers of buoyancy and gravity in the WEC-Sim model are defined as  [-2.79 ; 0 ;-1.5] and [-2.79; 0; 0.5] respectively, this creates this torque and instability you get in your simulations: 

According to the STL file provided the Cg and the Cb locations are shown in the fig below:
[image]
blue dot: Cb [m]	-3.124415, 0.55321582, -0.8874635
black dot: Cg [m]	0, 0, 0

2. Note that defining the Cg or Cb in the wecSimInputFile ``body(1).centerGravity = [-3.1 0.5 -1];`` will not solve the problem because WEC-Sim will ignore this inputs and use the values defined in the h5 file.

3. The mass of the buoy is defined as 8000 kgs, this is too low based on the buoy size, I recommend using `body(1).mass = 'equilibrium';   % kg`   

4. The hydrodynamic stiffness generated by WAMIT is small, for example the heave stiffness is 0.11666052, I think you need to rerun WAMIT after correcting the Cg location.  

5. Is the STL file provided has the full geometry or just the submerged part?

madewelr, 2024-07-02:
Apologies for the long delay. I was on a trip.

While I'm aware of the instability caused by the difference in the CG and CB placement, I can't figure out how to fix the issue. Any time I shift the CG placement in the WAMIT .pot file, it shifts the CG and the CB by the same amount, thus maintaining the offset torque issue. I'm not sure show to correct the CG location.

For the mass, I can set it to equilibrium, but the issue with that is that it calculates the mass as around 170,000kg which the mass should be around 10,000kg. I will leave it as equilibrium until the CG/CB issue is resolved and then adjust the mass.

The file attached here should have the geometry cut at the waterline. 
[OS.zip](https://github.com/user-attachments/files/16062686/OS.zip)

 I've played with both and originally provided the full geometry. However, thus far it still has the CG/CB issue regardless of the STL file.

madewelr, 2024-07-02:
I've been able to place the origin in the attached run such that the CG and CB are at 0,0,-0.3 and 0,0,-0.1 respectively. This seems to have greatly stabilized the run. Though it still errored out at around 55sec. Please see the run here: 
[OS.zip](https://github.com/user-attachments/files/16072753/OS.zip)

I was only able to place the CG under the CB by modifying the output file from WAMIT before I send it through BEMIO. I have to manually shift the location of the CB vertically by about 0.7m and in the x direction by about 0.000066m. 
If I leave the x unchanged, it errors in about 15seconds, so the X offset is important even with its tiny magnitude.

However, the hydrodynamic stiffness is a 6x6 matrix of zeros, which is an issue though one I am unsure how to go about solving.

madewelr, 2024-07-03:
@MShabara

I've been able to place the origin in the attached run such that the CG is in its natural location relative to the Rhino7 origin at 0,0,-0.518 and the CG is below that at 0,0,-1. However, the run still has the hydrodynamic stiffness of a 6x6 zeros. Please see the run here:
[OS.zip](https://github.com/user-attachments/files/16087761/OS.zip)

Additionally, a new error is now occurring:
image

I am unsure of how to solve the stiffness matrix issue as WAMIT is being run such that the CB and CG are correctly placed relative to each other, thus I imagine that there should be a stiffness while we clearly see there isn't.

I am also unsure how to solve the degenerate mass distribution issue. That is not one I understand and I can't find anything helpful in the other bug submissions.

I'm attaching the SOLIDWORKS, Rhino7, WAMIT GDF, and the STL files here:
[nomad.zip](https://github.com/user-attachments/files/16087828/nomad.zip)

madewelr, 2024-07-10:
@MShabara 

I've tried permutations the location of the CG relative to the CB of plus or minus 0.5m and 1m in both the X and Z plane. None of these variations created a stable run. Each failed at 0.0 seconds with the error:
[image]

Additionally, the stiffness matrix continues to be extremely low to zero, which I'm assuming is another symptom of whatever my problem is.

MShabara, 2024-07-10:
Hi @madewelr 

I was reviewing your BIMIO plots and noticed some irregular frequencies in the results. This suggests there might be an issue with your mesh. The mesh you provided appears as follows:

[image]

I modified the mesh using Cubit, and the resulting mesh looks like this:
[image]

I also did some comparison between your WAMIT results and a model Capytaine [Capytaine.zip](https://github.com/user-attachments/files/16169016/Capytaine.zip)
589678/3e560d52-b224-495c-9f6d-6b648c0304ee). The blue and orange curves are related WAMIT and Capytaine respectively. The effect in mesh quality can be observed in the figure below: 
[image]

Please do mesh refinement and mesh dependence study before using BEM solvers results in a WEC-Sim model.

I hope this fixes the instability problem.

madewelr, 2024-07-13:
@MShabara 

Using the mesh you provided did not produce viable results. 
I tried running the mesh through WAMIT to create the h5 file and that ran for approximately 30sec before erroring out. The body spun around for about 20 seconds before the error.
I also tried using the h5 file provided in your zip folder and that errored out immediately.

I also noticed that in both my mesh and your mesh, the CG and CB are inverted from where they should be. Do you have any idea how to flip those? I have been unsuccessful in doing so via WAMIT despite various attempts.

There is also the additional issue that I have that the Solidworks file is proprietary so I do not have access to it. The mesh file I provided is what I have to work with, so only minor modifications have been possible thus far (such as moving/rotating the entire mesh).

MShabara, 2024-07-14:
@madewelr 
The mesh provided produces viable hydrodynamic results based on the center of gravity (C.G) specified in the file you uploaded. I was able to get the WEC-Sim model to work by using some assumed C.G locations that satisfied the metacenter stability condition. Here are my recommendations to stabilize your model:

1. BEMrosetta:
[BEMrosetta](https://github.com/BEMRosetta/BEMRosetta/releases) is a useful, free, open-source tool that can help you adjust the C.G location and observe its effect on the hydrostatic matrix. Load your .gdf file and follow the red boxes shown in the figure below. Ensure that your hydrostatic stiffness values are non-negative.
[image]

2. Hydrodynamic solvers like Capytaine and NEMOH allow you to set the C.G. location before running the simulation. However, the center of buoyancy (C.B.) is calculated automatically in the solution process. I am not sure how WAMIT handles it, but am expecting that WAMIT have the same features.
 
3. Time Step Adjustment:
The time step you are using in the WEC-Sim simulation is large `simu.dt =0.1`. Try using smaller time steps or setting `simu.solver = 'ode45'` to let Matlab control the time step size.

closed 2024-07-14 by MShabara