WEC-Sim/WEC-Sim #1268: [Theory or Implementation]

riasatmorshed, 2024-06-27:
**Is your question request related to a problem? Please describe.**
Instead of using the regular wave, I would like my wave input to be Stokes' First Order Wave. That is to say, instead of using the default regular wave, I have written a code that calculates the elevation of the wave, and I would like to use that wave profile to my simulation. 
**Describe the theory or implementation approach that you have a question**
In my code, I am generating the <code>elevationData.mat</code> file and using the following lines:
<code>
waves = waveClass('elevationImport');
waves.elevationFile = 'elevationData.mat';
</code>
instead of the following lines:
<code>
waves = waveClass('regular');                            
waves.period = 2;                          % Wave Period [s]
waves.height = 0.02;                          % Wave Height [m]
</code>

**Personal project context**
I am trying to validate one of the CFD-run. It is a lab-scale OSWEC whose dimension is (0.2m x 0.1m x 0.8m) and 0.1m of freeboard. I used capytaine for generating the hydrodynamic data and then used BEMIO to create a .h5 file and then used WECSim. The CFD result is the OSWEC's position ranges from +3 degrees to -3 degrees whereas using <code>"regular wave"</code>, my output is +4 degrees to -4 degrees (around that range. not exactly 4 though). Later, I realized that since the wave input is not the same, that might be causing the discrepancy.  I am uploading a file for your reference that contains the WECSim project file along with the capytaine script. However, I would like to mention that, in the WECSim project file, some of the files may be irrelevant/redundant which I would like to request to ignore. 
**Describe the type of conclusion or resolution you are looking for**
I am actually looking for the manual, or step-by-step explanation on how WECSim handles this user-defined elevation or spectrum. I mean, I did what I could to my knowledge but I would assume there is a lot more to do since apparently they all are interconnected and my elevation code needs modifications as per the structure of the dependencies.  

**Additional context**
In the .zip file I am uploading, there is a <strong>OutputSnapshots</strong> folder to show what my output looks like. In addition to that, in one of the pictures, I have outlined a red rectangular box to show that these are something I didn't get for typical WECSim-run. I would be grateful if someone can walk me through how they are inter-related to better understand the system. I have already watched the waveClass and bodyClass tutorial vides and have been using WECSim for quite a some time. Thanks in advance.

[OSWEC_minge.zip](https://github.com/user-attachments/files/16006375/OSWEC_minge.zip)

riasatmorshed, 2024-07-01:
I am sorry for the mistake of not appropriately titling the issue. Looking forward to hearing from the community! Thanks!

dforbush2, 2024-07-03:
hi @riasatmorshed 

great questions. I'll try to give some more details and suggestions on the use of userDefinedElevation. From our docs: http://wec-sim.github.io/WEC-Sim/main/user/code_structure.html#elevationimport 

"The user-defined wave surface elevation must be defined with time (s) in the first column, and the wave elevation (m) in the second column"

This is elevation AT A POINT. And this elevation will be what is applied to your WECs excitation coefficients (i.e., N or N-m excitation force or torque per meter of wave height) So unless you set up your BEM run in a very particular and atypical way, this point also be your WEC location. 

Many simulation features cannot be used with userDefinedElevation, and it is worth considering if building an equivalent spectrum is a better approach, as per http://wec-sim.github.io/WEC-Sim/main/user/code_structure.html#elevationimport, last paragraph. This tends to be a more robust simulation option and one I would recommend in general. 

I'm a bit confused by your attachments...did you have a specific question related to the output structure? One thing I do notice is that your input elevation data is very nearly monochromatic: this means your results using elevationImport depend heavily on the discretization and range of your BEM data, which is probably undesirable. You can dig into waveClass code for exactly why this is, but basically it comes down to that your BEM data is all frequency domain, so to use elevationImport we have to go time --> frequency --> time for WECSim simulation vs. frequency --> time for other solution methods (e.g. spectrumImport). If you're encountering oddities in your output using elevationImport for this nearly-monochromatic time-series I strongly recommend using an equivalent spectrumImport.

With respect to WEC-Sim over-estimating the amplitude of motion for regular wave excitation vs. CFD, this is not unexpected. Unless you tell it otherwise, WEC-Sim is using the BEM excitation coefficients estimated from potential flow, whereas I presume your CFD uses a more realistic set of assumptions. Typically, to better match CFD/experimental results, we tune WEC-Sim simulations by adding viscous losses approximated as linear damping, quadratic drag, or Morison elements. From your wecSimInputFile, it does not appear you have done this. I would recommend this to improve simulation agreement.

dforbush2, 2024-07-11:
hi @riasatmorshed 

does this answer your question?

riasatmorshed, 2024-07-12:
@dforbush2, sorry I was travelling. That's why I didn't test it out, but it surely answers my question. I'll let you know here once I am done trying what you suggested. If you want, you can close the issue and I will reopen if I found difficulty implementing what you suggested on my own. Thanks for your response! I appreciate it very much!

closed 2024-07-17 by dforbush2