WEC-Sim/WEC-Sim #1271: Error in combineBEM/readWAMIT

Abby-McConnell, 2024-07-02:
**Describe the bug**
_I am trying to combine two .out files from WAMIT using readWAMIT and combineBEM. However, when I run the code, I get an error about "scalar structure required for this assignment". It does not matter which .out file comes first, the second one always errors out._

**WEC-Sim file(s)**
_I believe the error is in either readWAMIT or combineBEM._
_The two .out files are attached below. The section of code copied below is what generates the error:_
 
close all; clear; clc
% first call BEMIO
hydro = struct();
hydro = readWAMIT(hydro,'float - copy.out',[]);
hydro = readWAMIT(hydro,'LUPA_spar_D1_14m - copy.out',[]);
hydro = combineBEM(hydro);
dw=mean(diff(hydro.w));

[LUPA_spar_D1_14m - copy.txt](https://github.com/user-attachments/files/16071509/LUPA_spar_D1_14m.-.copy.txt)
[float - copy.txt](https://github.com/user-attachments/files/16071510/float.-.copy.txt)

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Paste the above code into a new window._
2. _Add the two .out files into the same folder as the new MATLAB code. You may need to re-save the two .txt files that are attached here to .out files by simply opening them in notepad, clicking 'Save As', and adding a .out to the end of the filename._
3. _Ensure that the filenames match those in the code._
4. _Run the program from the command line._

**Expected behavior**
_The code should combine the two .out files into one .h5 file. It has worked for older versions of bemio and readWAMIT, but no longer works._

**Screenshots**
_The error can be seen below. Note that the .out file is not the issue as the order of the two can be switched and the second file is always the one that throws the error. 

[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10/Windows 7 (attempted on two different workstations)
 - MATLAB R2023b and MATLAB R2021b
 - WECSim v6.0

**Additional context**
_N/A_

dforbush2, 2024-07-22:
Hi @Abby-McConnell 

Apologies for the delayed response. 

Interesting. On a fresh clone, my run also fails but because I get an 
``
Error: Inconsistent water depth

Error in bemio (line 5)
hydro = combineBEM(hydro);
''
By setting my tol = 10 % meters, I am able to get this to run. And it runs to completion without error and the output hydrostructure looks correct.  

This suggests that there is perhaps a version issue, and I would suggest that you update your local WEC-Sim clone and see it this works. 

Two things to be aware of: 
1) you do have different water depth by ~8cm. Likely not going to affect anything but it is strange you are not seeing the same error. 
2) Because these two BEM runs are each single-body, the non-block diagonal elements of the radiation forcing terms are all zero. If you anticipate significant cross-coupling between these body motions I suggest running a 2-body simulation. In any case, the magnitudes of the off-block-diagonal terms in this case will inform you of how reasonably they can be neglected.

Abby-McConnell, 2024-07-22:
Hi @dforbush2,

Thanks for the response! That could definitely be the case, I was able to code a workaround that worked with the clone I'm using, but an update is probably in order it seems. 

Thanks for the suggestions, I'll do some 2 body runs and make certain that everything is the same depth.

closed 2024-07-22 by Abby-McConnell