WEC-Sim/WEC-Sim #1276: WEC-Sim Example RM3

donprofaghatise, 2024-07-13:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._
I am a new user and was trying to follow the step by step guide for example RM3
**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._

I updated the .h5 using Bimo but when I run wecSim I get error message:

 
**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

**Expected behavior**
_A clear and concise description of what you expected to happen._

I expected to see plots from the simulations but it ran to this extent (see below) and then produced an error message
(Extent it ran )
Initializing the Simulation Class...
	Case Dir: C:\WEC-Sim-main\examples\RM3 
Elapsed time is 2.087668 seconds.

WEC-Sim Pre-processing ...   
	Infinite water depth specified in BEM and "waves.waterDepth" not specified in input file.
Set water depth to 200m for visualization.
Elapsed time is 0.167959 seconds.

WEC-Sim Simulation Settings:
	Start Time                     (sec) = 0
	End Time                       (sec) = 400
	Time Step Size                 (sec) = 0.1
	Ramp Function Time             (sec) = 100
	 Number of Time Steps     = 4000 

Wave Environment: 
	Wave Type                            = Regular Waves (Constant Hydrodynamic Coefficients)
	Wave Height, height (m)                   = 2.5
	Wave Period, period (sec)                 = 8

List of Body: Number of Bodies = 2 

	***** Body Number 1, Name: float *****
	Body CG                          (m) = [0,0,-0.72]
	Body Mass                       (kg) = 725833 
	Body Moments of Inertia       (kgm2) = [2.09073E+07,2.13061E+07,3.70855E+07]
	Body Products of Inertia      (kgm2) = [0,0,0]

	***** Body Number 2, Name: spar *****
	Body CG                          (m) = [0,0,-21.29]
	Body Mass                       (kg) = 886687 
	Body Moments of Inertia       (kgm2) = [9.44196E+07,9.44071E+07,2.85422E+07]
	Body Products of Inertia      (kgm2) = [0,0,0]

List of PTO(s): Number of PTOs = 1 

	***** PTO Name: PTO1 *****
	PTO Stiffness           (N/m;Nm/rad) = 0
	PTO Damping           (Ns/m;Nsm/rad) = 1.2E+06

List of Constraint(s): Number of Constraints = 1 

	***** Constraint Name: Constraint1 *****

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

 (see below error message):
Simulating the WEC device defined in the SimMechanics model C:\WEC-Sim-main\examples\RM3\RM3.slx...   
Error using simulationClass/loadSimMechModel
block_diagram does not have a parameter named 'SimMechanicsOpenEditorOnUpdate'

Error in initializeWecSim (line 533)
simu.loadSimMechModel(simu.simMechanicsFile);

Error in run (line 99)
evalin('caller', strcat(script, ';'));

Error in wecSim (line 43)
run('initializeWecSim'); - Show complete stack trace

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]  WINDOWS 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]   2022b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]
 Not too sure of the version but download it on 13 July 2024 (today).

**Additional context**
_Add any other context about the problem here._

dforbush2, 2024-07-22:
This is usually a MATLAB version issue when it cannot open a block diagram created with a newer version of Simscape Multibody and/or simulink. I am unable to reproduce this error. 

Did you clone from the most recent version of the repository (https://github.com/WEC-Sim/WEC-Sim , big green button) or download from a static code release https://github.com/WEC-Sim/WEC-Sim/releases? Also, are you calling >>wecSim from the command line or are you attempting to run from simulink?

I do notice the constraints library is 2022b. This should not be a problem but it is inconsistent with other library versions. A thing you can try is to revert this version to 2020b locally by:
1) navigating to your WEC-Sim source directory
2) >> bdclose all
3) >> removeWecSimSource, then navigate to source/lib/wec-sim
4) Rename source/lib/wec-sim/WECSim_Lib_Constraints.slx to source/lib/wec-sim/WECSim_Lib_Constraints_org.slx (or something, just needs to be different),
5) open the file renamed in 4). 
6) Export to previous version (Matlab 2020b .slx), and name 'WECSim_Lib_Constraints.slx'. Then bdclose all again.
7) navigate back to WEC-Sim, then run addWecSim source.

Retry running your example, and let me know if this fixes it!

dforbush2, 2024-07-22:
Also please see https://github.com/WEC-Sim/WEC-Sim/issues/587

may be as simple as a simulink restart

dforbush2, 2024-07-31:
did this resolve your issue?

donprofaghatise, 2024-08-02:
Good morning,

Thank you so much for this. I have taken the steps you mentioned in your
earlier email but was still met with the same problem. Below is the
error message I got  after modification (as advised). I truly appreciate
your help and if there is any more support I can get , I truly will
appreciate it.
 Below is what I get running from the command line Example RM3.
>> wecSim
WEC-Sim: An open-source code for simulating wave energy converters
Version: 6.0

Initializing the Simulation Class...
Case Dir: C:\WEC-Sim-1-master\examples\RM3

WEC-Sim Input From Standard wecSimInputFile.m Of Case Directory...
WEC-Sim: An open-source code for simulating wave energy converters
Version: 6.0

Initializing the Simulation Class...
Case Dir: C:\WEC-Sim-1-master\examples\RM3
Elapsed time is 0.408921 seconds.

WEC-Sim Pre-processing ...
Infinite water depth specified in BEM and "waves.waterDepth" not specified
in input file.
Set water depth to 200m for visualization.
Elapsed time is 0.344489 seconds.

WEC-Sim Simulation Settings:
Start Time                     (sec) = 0
End Time                       (sec) = 400
Time Step Size                 (sec) = 0.1
Ramp Function Time             (sec) = 100
Number of Time Steps     = 4000

Wave Environment:
Wave Type                            = Regular Waves (Constant Hydrodynamic
Coefficients)
Wave Height, height (m)                   = 2.5
Wave Period, period (sec)                 = 8

List of Body: Number of Bodies = 2

***** Body Number 1, Name: float *****
Body CG                          (m) = [0,0,-0.72]
Body Mass                       (kg) = 725833
Body Moments of Inertia       (kgm2) = [2.09073E+07,2.13061E+07,3.70855E+07]
Body Products of Inertia      (kgm2) = [0,0,0]

***** Body Number 2, Name: spar *****
Body CG                          (m) = [0,0,-21.29]
Body Mass                       (kg) = 886687
Body Moments of Inertia       (kgm2) = [9.44196E+07,9.44071E+07,2.85422E+07]
Body Products of Inertia      (kgm2) = [0,0,0]

List of PTO(s): Number of PTOs = 1

***** PTO Name: PTO1 *****
PTO Stiffness           (N/m;Nm/rad) = 0
PTO Damping           (Ns/m;Nsm/rad) = 1.2E+06

List of Constraint(s): Number of Constraints = 1

***** Constraint Name: Constraint1 *****

Simulating the WEC device defined in the SimMechanics model
C:\WEC-Sim-1-master\examples\RM3\RM3.slx...
Error using simulationClass/loadSimMechModel (line 218)
block_diagram does not have a parameter named
'SimMechanicsOpenEditorOnUpdate'

Error in initializeWecSim (line 533)
simu.loadSimMechModel(simu.simMechanicsFile);

Error in run (line 112)
evalin('caller', strcat(scriptStem, ';'));

Error in wecSim (line 43)
run('initializeWecSim'); - Show complete stack trace

dforbush2, 2024-08-02:
Hello @donprofaghatise

I am unable to reproduce your error on my end and I think it is likely a version issue. To help me debug further, can you please type

`ver`

into the command line and paste the output of that command into your reply?

Best,
Dom

donprofaghatise, 2024-08-02:
Thank you so much for the support.  See below - as instructed.

>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 24.1.0.2578822 (R2024a) Update 2
MATLAB License Number: 40469364
Operating System: Microsoft Windows 10 Home Version 10.0 (Build 19045)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM)
64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 24.1
 (R2024a)
Simulink                                              Version 24.1
 (R2024a)
Computer Vision Toolbox                               Version 24.1
 (R2024a)
Deep Learning Toolbox                                 Version 24.1
 (R2024a)
Global Optimization Toolbox                           Version 24.1
 (R2024a)
Image Processing Toolbox                              Version 24.1
 (R2024a)
Optimization Toolbox                                  Version 24.1
 (R2024a)
Reinforcement Learning Toolbox                        Version 24.1
 (R2024a)
Simulink Coverage                                     Version 24.1
 (R2024a)
Statistics and Machine Learning Toolbox               Version 24.1
 (R2024a)

dforbush2, 2024-08-02:
Ah I see the problem. Unfortunately you'll need the Simscape multibody toolbox to utilize WEC-Sim.

http://wec-sim.github.io/WEC-Sim/main/user/getting_started.html#matlab-requirements

closed 2024-08-05 by dforbush2