WEC-Sim/WEC-Sim #1277: [WEC-Sim Applications] RM3 Power Output

sherlocksilver, 2024-07-13:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
The WEC-Sim application case in question is RM3.

**Describe the bug or issue**
I have successfully run a simulation using the RM3 case and obtained the Mechanical absorption power. I would like to know how to determine the duration for which this power can be absorbed.

**To Reproduce**
Steps to reproduce the behavior:

Open the WEC-Sim application case RM3.
Run the simulation.
Note the Mechanical absorption power from the simulation results.
Request information on how to calculate the duration for which this power can be absorbed.

**Expected behavior**
N/A

**Screenshots**
N/A

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2021b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v6.0]

**Additional context**
N/A

dforbush2, 2024-07-22:
Hi @sherlocksilver 

The mechanical power for the RM3 example $/examples/RM3 is calculated as P_{mech} = (-pto(1).stiffness * pto(1).position + -pto(1).damping * pto(1).velocity) * pto(1).velocity

And is a function of the user-defined PTO stiffness and damping parameters. Assuming these are constant, then the relative displacement of the bodies is what affects the result and this is driven by the Cummins equation on each body. The short answer to your question is: as long as the spectral characteristics of the excitation forces remain unchanged, the _average_ mechanical power capture will also remain the same. 

The longer answer depends on if we 're discussing a simulation or in-the-field context. For simplicity, consider that you have a regular wave: the power at each corresponding instant on any wave or set of waves will be identical. However, the average power will only be identical if you are averaging over an integer number of wave periods. If you are averaging over many waves, this difference will become very small, but over a small number of waves this can be significant. Thus, it is generally important that any performance statistic is evaluated over _an integer number of wave periods_. 

Now consider the more complex example of an irregular wave. The basic argument holds: we want to average over an integer number of wave repetitions, but in this case, the wave repetition is not just one wave peak period but the time-domain representation of the irregular wave spectra. In simulation, using the irregular wave class, this by default is 1/(simulation time): thus running multiple simulations each with randomized wave phase will be expected to create identical statistics. You can create custom spectra with multiple wave repetitions in a single simulation if desired. 

Of course, in the field, waves never perfectly repeat and so we can't get "an integer number of wave periods", so in this case, we just need to make sure that the period over which we average is 1) long enough so that the wave-by-wave variation is minimized and 2) short enough that the spectral description of the sea-state can be approximated as constant. There is not a single right answer on how long this should be, but there are standards with respect to WEC performance characterization that provide robust guidance https://webstore.iec.ch/en/publication/22593

This is a complex question, but hopefully this response offers some insight.

sherlocksilver, 2024-07-29:
Hello @dforbush2 , 

Thanks for your amazing reply. I have another basic question.

I have a set of scenarios with varying significant wave heights (Hs) and periods. For example, I have the following cases:

1. Hs = 2m, T = 7s
2. Hs = 2.5m, T = 7s
3. Hs = 3m, T = 7s
4. Hs = 2m, T = 7.25s
5. Hs = 2.5m, T = 7.25s
and so on...

For each case, I update the wave conditions in my wecSimInputFile.m, and run a new simulation to obtain the mechanical power (P1, P2, P3, P4, P5, etc.). in W or kW.
How can I determine the average power of this WEC across all these different sea states? Is there any appropriate method to calculate the average power?

Thank you in advance!

Regards

dforbush2, 2024-07-31:
Hi @sherlocksilver 

Great question. Again the "right" answer is in the wave resource characterization standards https://webstore.iec.ch/en/publication/22593. I highly recommend a direct review of the standards if able but I will summarize here. 

Typically when a wave resource at a site is characterized, a "joint probability distribution" is developed. This is a 3D representation of Probability of occurrence (on Z axis) as a function of the significant wave height (on X or Y) and wave peak or energy period (on Y or X). The sum of all the probabilities (all bins of wave periods and wave heights) should be 1. This joint probability distribution, or JPD, depends only on the site. 

For each individual WEC device, there is a separate matrix relating power capture (on the Z axis) as a function of the significant wave height (on X or Y) and wave peak or energy period (on Y or X). This power matrix depends only on the WEC device. Performing an element-wise multiplication of this power matrix with the JPD and then integrating over X and Y will be an estimate of the average power that device will make in that site.

sherlocksilver, 2024-08-03:
Hi @dforbush2 ,

Thank you for your incredibly detailed and insightful reply! I appreciate your guidance on using the joint probability distribution and power matrix for estimating average power. I have a couple more questions, though they might be quite basic:

Details on RM3 and OSWEC:
Could you provide more information on the RM3 and OSWEC devices, specifically regarding their material, density, internal structure (e.g., are the buoy and OSWEC hollow?), and the boundary conditions used in the hydrodynamics solver? I'm trying to get a comprehensive understanding of these devices.

Interpreting 'hydro' Figures:
After running BEMIO, I've encountered various figures such as Normalized Added Mass, Normalized Radiation Damping, Normalized Radiation Impulse Response Functions, Normalized Excitation Force Magnitude, Excitation Force Phase, and Normalized Excitation Impulse Response Functions. How do I determine which graph pattern is better for harnessing wave energy? For instance, if we compare different WECs and the Normalized Added Mass value is higher than others, is that preferable? What kind of decisions can we make from these plots? For example, if we get Normalized Radiation Damping for two different sizes of the same converter in the same sea state and find one has a higher damping value, does that indicate it is a better converter? Similarly, how should we interpret the other figures from a general perspective?

I know these questions might be quite basic, but if possible, I would greatly appreciate your super-duper reply. Your insights are particularly valuable to me, and your detailed response would be most helpful in guiding my understanding. Alternatively, if there are specific books or lectures you would recommend, I would be grateful for those resources as well.

Thank you again for your help!

Regards.

dforbush2, 2024-08-05:
With respect to reference model device information, that is freely available here: https://energy.sandia.gov/programs/renewable-energy/water-power/projects/reference-model-project-rmp/

With respect to optimal WEC design, this paper and its successors will offer you a lot of guidance. https://www.sciencedirect.com/science/article/pii/S1364032121000861

In your research be wary that many different things in the literature are referred to as "co-design". If you follow the work of the authors' of the suggested paper (among others), what is referred to as "co-design" is quantitative, specific, and rigorously formulated. You'll see that there is not really a simple answer to your question, and that one cannot consider hydrodynamic parameters of WECs in isolation.

dforbush2, 2024-08-05:
Since your question has been answered, I'm going to close this issue so that future unique questions are more easily searchable for all users.

closed 2024-08-05 by dforbush2