WEC-Sim/WEC-Sim #1278: [WEC-Sim Applications] Question about PTO-Sim example of OSWEC

fanofdw, 2024-07-23:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
[https://github.com/WEC-Sim/WEC-Sim_Applications/tree/main/PTO-Sim/OSWEC/OSWEC_Hydraulic_PTO](url)

**Describe the bug or issue**
Hello, WEC-Sim team. I have run the PTO-Sim/OSWEC/OSWEC_Hydraulic_PTO example code, and I found that this example is not converged. The piston pressure, PTO force and electrical power (U*I) present a gradually expanding trend.

**To Reproduce**
1. Add the code below into userDefinedFunctions.m.
`figure();
plot(output.ptoSim(7).time,output.ptoSim(7).voltage.*output.ptoSim(7).current/1000);
set(findall(gcf,'type','axes'),'fontsize',16);
xlabel('Time (s)');
ylabel('Power (kW)');
title('Power');
grid on;`

2. generate .h5 file by the bemio.m and oswec.out files from [https://github.com/WEC-Sim/WEC-Sim/tree/main/examples/OSWEC/hydroData](url)

3. Run wecSim and check results.

**Expected behavior**
The piston pressure, PTO force and electrical power (U*I) should converge.

**Screenshots**
[image]
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) R2021b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v6.0

**Additional context**

fanofdw, 2024-07-23:
In addition, I studied the Simulink blocks OSWEC_Hydraulic_PTO carefully, and I might found two bugs (please see below figures). 
[image]
[image]

jleonqu, 2024-07-25:
Hello @fanofdw 

Thanks for opening this issue.

The results for this example will be affected by the parameters of the hydraulic system that you are using. I would suggest to review the parameters that you are using to make sure that they are correct for the power ratings of the OSWEC example. For example, if the displacement in the hydraulic motor is too small, the fluid won't have where to go and the pressure will continuously go up. In addition to this, the gradually increasing pressure could be a consequence of a small accumulator, or a low shaft speed in the electric motor, or a check valve that is not correctly sized for the application. You should change these values according to your specific application.

Regarding the "Rotary to Linear Adjustable Rod", these two issues have information on how the equations were derived: #247 and #993

fanofdw, 2024-07-29:
Hello @jleonqu ! Thank you so much for your rapid and detailed reply. You're right that the results for this example are affected by the parameters of the hydraulic system. I made corrections to these parameters, and I can now achieve stable pressure. Thanks again, and I hope you have a lovely day.

closed 2024-08-03 by fanofdw