WEC-Sim/WEC-Sim #1279: [Developer Issue]: Aligning wave spread function with IEC standards

dforbush2, 2024-07-24:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
Wave spreading function

**Describe the issue**
IEC 62600 standards for wave resource assessments defines a wave spreading parameter that is frequency-dependent. For users who desire to simulate a site characterized according to this wave resource dependency, it doesn't seem that it is straightforward to utilize this information in WEC-Sim, which uses a uniform spread over all frequencies of a spectra that is discretized into a user defined set of directions. I suggest such guidance or functionality to directly use this sort of characterization is developed. 

If anyone has done this or has information on how to do this within the current WEC-Sim implementation, it would be appreciated.

dforbush2, 2024-08-02:
I have created a function to take in IEC characterizations into WEC-Sim's present wave implementation. It will be a good approximation if the excitation bandwidth is clearly defined, mostly unidirectional, and with a fairly consistent spread over excited frequencies. OR if the above criteria can be met using a relatively small number of wave trains (i.e., the super position of swell and local wind waves that result in 2 distinct bandwidths of excitation, in 2 primary directions, with 2 fairly consistent spreads can be well-approximated in this way using 2 wave trains). 

In progress on a fully frequency-resolved calculation method.

kmruehl, 2024-10-30:
Refer to PRs https://github.com/WEC-Sim/WEC-Sim/pull/1332 and https://github.com/WEC-Sim/WEC-Sim/pull/1328

closed 2025-02-05 by dforbush2