WEC-Sim/WEC-Sim #1282: Inquiry on OWC Simulation in WEC-Sim

samir-miloud, 2024-07-28:
Hello everyone,

I'm currently working on a project involving an Oscillating Water Column (OWC) device and am wondering if WEC-Sim can be used to simulate this type of technology. If so, could you please point me towards any known applications or examples that use WEC-Sim to model an OWC?
Thank you in advance for your assistance!

Best regards,
Samir

jleonqu, 2024-07-29:
Hello @samir-miloud 

A friendly request: can you reopen your issue using the suggested template?

Thanks! Looking forward to help you in the new issue.

dforbush2, 2024-07-31:
Hi @samir-miloud and others

We had an offline correspondence and solved this issue.

1) the wecSim applications repository https://github.com/WEC-Sim/WEC-Sim_Applications has an OWC example (among lots of others), so that is recommended to all users as a starting point.
2) the OWC example in the wec-Sim_Applications repository claims to require the control systems toolbox, but I actually have already implemented a work-around there.

 On lines 46 through 49 of wecSimInputFile.m
`lpf50 = tf(2*pi*50,[1 2*pi*50]);
lpfNum = 2*pi*50;
lpfDen = [1 2*pi*50];
LPF50 = [lpf50 0 0 0 0 0 0; 0 lpf50 0 0 0 0 0; 0 0 lpf50 0 0 0 0; 0 0 0 lpf50 0 0 0; ...
    0 0 0 0 lpf50 0 0; 0 0 0 0 0 lpf50 0; 0 0 0 0 0 0 lpf50];`

Lines 46 and 49 can actually be commented out. The tf() call from Matlab requires the Control Systems toolbox, but the transfer function block in simulink, informed by lines 47 and 48, does not. The reason for this low pass filter is that is breaks the algebraic loop in lieu of the transport delay block in the body block library and allows for faster simulation in this case. 

[image]

I will leave this issue open until I make a PR to the applications repository removing the extraneous lines and toolbox dependencies

Best,
Dom

samir-miloud, 2024-08-01:
Hello,

Following your previous assistance, I commented out lines 46 to 49 in the wecSimInputFile.m file as you suggested. However, I am now encountering a new error when running the Simulink model GBM1.slx in the test case located in /Documents/GitHub/WECSIM/WEC-Sim_Applications/OWC/OrificeModel.

Below is the summary of the test case execution as well as the figure of the Simulink block diagram GBM1.slx:

**Execution Summary:**

>> wecSim
WEC-Sim: An open-source code for simulating wave energy converters
Version: 5.0.1

Initializing the Simulation Class...
	Case Dir: /home/samir/Documents/GitHub/WECSIM/WEC-Sim_Applications/OWC/OrificeModel 

WEC-Sim Input From Standard wecSimInputFile.m Of Case Directory... 
WEC-Sim: An open-source code for simulating wave energy converters
Version: 5.0.1

Initializing the Simulation Class...
	Case Dir: /home/samir/Documents/GitHub/WECSIM/WEC-Sim_Applications/OWC/OrificeModel 
Warning: Center of gravity and center of buoyancy are overwritten by h5 data for hydro bodies. 
> In bodyClass/checkInputs (line 186)
In initializeWecSim (line 140)
In run (line 91)
In wecSim (line 43) 
Elapsed time is 12.506987 seconds.

WEC-Sim Pre-processing ...   
	Infinite water depth specified in BEM and "waves.waterDepth" not specified in input file.
Set water depth to 200m for visualization.
Elapsed time is 9.133524 seconds.

WEC-Sim Simulation Settings:
	Start Time                     (sec) = 0
	End Time                       (sec) = 130
	Time Step Size                 (sec) = 0.005
	Ramp Function Time             (sec) = 10
	Convolution Integral Interval  (sec) = 15
	 Number of Time Steps     = 26000 

Wave Environment: 
	Wave Type                            = Irregular Waves (Predefined Random Phase)
	Spectrum Type                        = Pierson-Moskowitz  
	Significant Wave Height, Hs      (m) = 1
	Peak Wave Period, Tp           (sec) = 4

List of Body: Number of Bodies = 1 

	***** Body Number 1, Name: test17a *****
	Body CG                          (m) = [0,0,0]
	Body Mass                       (kg) = 589.049 
	Body Diagonal MOI              (kgm2)= [99.28,11.04,99.2]

List of PTO(s): No PTO in the system

List of Constraint(s): Number of Constraints = 1 

	***** Constraint Name: Constraint1 *****

Simulating the WEC device defined in the SimMechanics model /home/samir/Documents/GitHub/WECSIM/WEC-Sim_Applications/OWC/OrificeModel/GBM1.slx...   
Elapsed time is 3.784086 seconds.
------Error in the Model, Error-Causes/Messsages are :
Error evaluating parameter 'Numerator' in 'GBM1/Flex Body/Morrison Element and Viscous Damping Force Calculation/Morison Element and Viscous Damping Force Calculation/Subsystem/Transfer Fcn1'
>> 

**Simulink Block Diagram Figure:**

[image]

Thank you in advance for your help in understanding and resolving this new issue.

Best regards,

Samir

dforbush2, 2024-08-01:
As per previous instructions, please comment out ONLY Lines 46 and 49. 

"Lines 46 and 49 (beginning lpf50 and LPF50) can actually be commented out"

dforbush2, 2024-08-01:
You have commented out 47 and 48 as well, which defines lpfNum and lpfDen. ONLY comment out 46 and 49.

------Error in the Model, Error-Causes/Messsages are :
Error evaluating parameter 'Numerator' in 'GBM1/Flex Body/Morrison Element and Viscous Damping Force Calculation/Morison Element and Viscous Damping Force Calculation/Subsystem/Transfer Fcn1'

You get this error because lpfNum is undefined. You will get another one if lpfDen is undefined. This is why you must uncomment lines 47 and 48.

samir-miloud, 2024-08-01:
Thank you for the instructions regarding the definition of the lpfNum and lpfDen parameters. I have followed your recommendations by uncommenting lines 47 and 48, and commenting only lines 46 and 49, as suggested.

[image]

However, despite these adjustments, the issue persists. I am still encountering difficulties with the model.
Could you please review the changes made and the simulation log file to help identify the source of the problem?
[simulation.log](https://github.com/user-attachments/files/16460908/simulation.log)

Thank you for your assistance.

Best regards,

dforbush2, 2024-08-01:
It appears that the parameter body.largeXYDisplacement.option is not being found. This is a feature introduced fairly recently. Is it possible that your version of WEC-Sim is out of date? Please pull the latest repository and retry your model. 

If this does not resolve your error, please provide a *zip file of your case directory after attempting to run this case.

samir-miloud, 2024-08-02:
Hello,
Thank you for your message. I have a few questions for clarification:
Are you asking me to update WEC-Sim to the latest version by cloning the repository and retrying my model?
You sent me a figure below. Could you explain why you sent it? Is it for making modifications to the WECSim_Lib_Body_Elements.slx file?

[image]
Regarding this, I haven't been able to make any modifications to the file because I am currently facing another issue with opening .slx files in Simulink. I am working on resolving this problem.

Thank you for your assistance.

Best regards,

Samir

dforbush2, 2024-08-02:
You should not have to make any modifications to the library blocks or the blocks of the model. I sent the image to indicate where in the model the lpfNum and lpfDen parameters that were causing the errors are called. I believe we have resolved this part of the issue. 

With regard to your most recent error, yes, I am suggesting you update your WecSim repository. The model is attempting to utilize a parameter in the bodyClass called largeXYDisplacement.option.

If your repository is not up-to-date, this parameter may not exist.

samir-miloud, 2024-08-03:
Dear Domnic,

Thank you for your assistance. I followed your instructions and cloned the latest version of WEC-Sim. I am pleased to inform you that the model is now working perfectly.
I appreciate your prompt and valuable response. Your guidance was invaluable in resolving this issue.
Attached, you will find a video of the OWC test case.
Thank you once again for your support.

Best regards,

Samir

[GBM1.zip](https://github.com/user-attachments/files/16478668/GBM1.zip)

closed 2024-08-05 by dforbush2

samir-miloud, 2024-08-08:
Hello everyone,

I’m currently searching for an application for an Oscillating Water Column (OWC) under WEC-Sim, specifically within the WEC-Sim/examples/BEMIO/CAPYTAINE directory. Unlike the other examples like coer_comp, cubes, cylinder, ellipsoid, oswec, rm3, and sphere, which handle systems with 6 degrees of freedom (DOF), the OWC involves 7 DOF, which is making my search more challenging.
I’m a bit stuck and going in circles trying to find an example or resource suited for a 7 DOF OWC model. If anyone has worked on something similar or can point me in the right direction, I would greatly appreciate your help.
Thank you in advance for any advice or information!

Best regards,

samir-miloud, 2024-08-09:
Hello,

I am looking for the gbm_dofs.py and call_capytaine.py scripts that would allow me to simulate the hydrodynamic problem of the spar buoy OWC. If anyone has these scripts or can guide me towards a solution, I would be grateful.

Thank you in advance for your help.

Best regards,

samir-miloud, 2024-08-10:
Hello,

I am currently working on the hydrodynamic modeling of an OWC (Oscillating Water Column) float and would like some assistance with using Capytaine in conjunction with the GBM (Generalized Body Modes) technique.

Thank you for your help!

Samir