WEC-Sim/WEC-Sim #1283: Run the RM3_cHydraulic_PTO example in the pto-sim[BUG]

RitchieWMJ, 2024-07-31:
**Describe the bug**
When I run the RM3_cHydraulic_PTO example,The following error has occurred:
Incorrect use of wecSim
Error calculating the Frames block (mask) "RM3_cHydraulic_PTO/Global Reference Frame" callback.
The callback string is 'run('stopwecSim')'
Cause :StopFcn' incorrectly uses the index at wecSim position 2 beyond the array boundary. The index cannot exceed 3

**WEC-Sim file(s)**
N/A

**To Reproduce**
N/A

**Expected behavior**
N/A

**Screenshots**
[image]

[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 11]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [r2022b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v5.0.1]

**Additional context**
N/A

jleonqu, 2024-08-06:
Hello @RitchieWMJ 
I am running the RM3_cHydraulic_PTO example with no issues using the same Matlab version that you are using. I suggest to pull the latest version of the WEC-Sim and WEC-Sim_Applications main branches, and then run the example again.

jleonqu, 2024-08-23:
Closing this issue due to inactivity.

closed 2024-08-23 by jleonqu