WEC-Sim/WEC-Sim #1291: OWC (7 DOF) Application in WEC-Sim BEMIO/CAPYTAINE

samir-miloud, 2024-08-08:
Hello everyone,

I’m currently searching for an application for an Oscillating Water Column (OWC) under WEC-Sim, specifically within the WEC-Sim/examples/BEMIO/CAPYTAINE directory. Unlike the other examples like coer_comp, cubes, cylinder, ellipsoid, oswec, rm3, and sphere, which handle systems with 6 degrees of freedom (DOF), the OWC involves 7 DOF, which is making my search more challenging.
I’m a bit stuck and going in circles trying to find an example or resource suited for a 7 DOF OWC model. If anyone has worked on something similar or can point me in the right direction, I would greatly appreciate your help.
Thank you in advance for any advice or information!

Best regards,

jtgrasb, 2024-08-09:
Please see the [WEC-Sim Applications Repository](https://github.com/WEC-Sim/WEC-Sim_Applications) which has an OWC example.

For all issues, we ask users to open GitHub issues using an appropriate template. This is helpful for the WEC-Sim team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general. Please edit this issue or open a new issue to use the appropriate issue template if you have additional questions. Thank you.

samir-miloud, 2024-08-09:
Hello,

I am looking for the gbm_dofs.py and call_capytaine.py scripts that would allow me to simulate the hydrodynamic problem of the spar buoy OWC. If anyone has these scripts or can guide me towards a solution, I would be grateful.

Thank you in advance for your help.

Best regards,

samir-miloud, 2024-08-10:
Hello,

I am currently working on the hydrodynamic modeling of an OWC (Oscillating Water Column) float and would like some assistance with using Capytaine in conjunction with the GBM (Generalized Body Modes) technique.

Thank you for your help!

Samir

dforbush2, 2024-08-14:
see #1293

closed 2024-08-14 by dforbush2