WEC-Sim/WEC-Sim #1299: [WEC-Sim Applications]

HareendraAmaratunge, 2024-08-23:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._

**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

HareendraAmaratunge, 2024-08-23:
I am unable to run the RM3 example file in the moorDyn located in the WEC-SIM_Applications. An error is occuring and no mechanics explorer is generated.

[image]

RyanDavies19, 2024-08-30:
Hi @HareendraAmaratunge,

Have you added all the files here: https://github.com/WEC-Sim/MoorDyn to the WEC-Sim/source/functions/moorDyn folder? The error you are getting makes me thing that the MoorDyn files are not in the proper location.

HareendraAmaratunge, 2024-09-02:
Hi @RyanDavies19,

Yes. I clone the whole https://github.com/WEC-Sim/MoorDyn repository to my
WEC-Sim/source/functions/moorDyn folder.

closed 2024-09-02 by HareendraAmaratunge