WEC-Sim/WEC-Sim #1300: [Theory or Implementation] Explanation Regarding the OSWEC with Surface-piercing

riasatmorshed, 2024-08-29:
**Is your question request related to a problem? Please describe.**
My question is about the clarification regarding the OSWEC implementation whose flap has surface piercing.

**Describe the theory or implementation approach that you have a question**
I am using Capytaine as a BEM solver. To my knowledge, we need to calculate hydrodynamic coefficients and hydrostatic stiffness for the submerged part of the flap only. However, say, I have a flap that has a surface piercing of 0.1meter and the total height of the flap is 0.8 meter. So, I did the hydrodynamic calculation for 0.7m which is the submerged part and in the Capytaine, I specified the center of gravity for the submerged part only.

For example, my flap is a rectangular uniform flap and for the whole body, the center of gravity is (0,0,-0.30m). But if I consider the submerged part only then the center of gravity becomes (0,0,-0.35m). And center of gravity changes hydrostatic stiffness. Am I right? In addition to that, to run the BEMIO function, I need to specify CB and CG in the "Hydrostatics" file and the hydrostatic stiffness in the "KH" file. What values should I put in- the values for the whole body or the values for the submerged part only? Just a clarification, when I am referring to the whole body-values, my origin still lies in the free surface water level.

That brings me to the second question. While I am inputting my values for mass and inertia in the wecsim input file, should I input mass for my whole body or for the submerged part only. Secondly, mine is a pitch device, so in the inertia matrix, should I use the [Iyy Iyy Iyy] or [Ixx Iyy Izz]? The reason I am asking this is because in the tutorial case, it probably uses [Iyy Iyy Iyy]. And should those Ixx/Iyy/Izz values should be calculated about the CG of the body or about the origin which lies at the free surface?

My third question is, to my understanding, the .stl file is only there for visualization purpose if I don't really care for nonlinear buoyancy or nonlinear FK force, therefore, does not really affect the time-domain simulation, right? But even if I add the .stl file, my .stl should have its origin at the CG of the body, right? and this is purely for the visualization purpose- am I correct?

**Personal project context**
I am just trying to simulate a lab-scale OSWEC and I sort of went about without caring much about the discrepancy between the whole body-submerged body thing. But my simulation resulted in a rotation angle of 30-40 degrees which made me think if I am doing something wrong because to my knowledge, the linear assumption breaks down if the rotation angle (position) goes beyond 30 degrees. Am I right?

**Describe the type of conclusion or resolution you are looking for**
I am sorry if the questions seem too trivial to answer and I looked into the WECSim videos and tried to scavenge answers from the already closed github issues, but can't seem to find a clear-cut answer to those questions. Thanks in advance.

**Additional context**
N/A

riasatmorshed, 2024-09-15:
I would like to add another question for the sake of clarification. While we are populating KH.dat file for BEMIO function, the hydrostatic stiffness matrix should be about the center of gravity, not for center of rotation, right?

I took the OSWEC tutorial case and looked into the KH.dat file of that tutorial case. What turned out is the value corresponds with the hydrostatic stiffness matrix calculated about the center of gravity. 

The reason I am asking is hydrostatic stiffness works as a restoring term and if so to account for the righting lever, shouldn't we calculate the stiffness matrix about the center of rotation, not for center of gravity? Or we don't do this because WECSim later convert everything to center of rotation from the CG?

dav-og, 2024-09-25:
Apologies for the delayed response,

You are correct: Capytaine only requires the wetted surface in order to compute the hydrodynamic coefficients. However, for use with WEC-Sim, you should compute these hydrodynamic coefficients about the CoG of the _complete_ body.

Similarly, to your second question: in WEC-Sim you should use the mass, inertia and CoG of the complete body. You should use [Ixx Iyy Izz] but if the body is constrained to only pitch then you shouldn't notice a difference compared to [Iyy Iyy Iyy] since there will be no rotations around x or z axis.

Thirdly, yes in your case the .stl will only be used for visualization (since you are not using the nonlinear hydrodynamic options). And yes, the stl should have (0,0,0) set at the body's CoG.

Lastly, yes the KH matrix should be calculated at CoG. WEC-Sim computes all the hydrodynamic forces at the CG and provides these to Simscape Multibody to simulate the system in the time-domain.

I hope that helps! It can be confusing at first to ensure consistent of CoG etc across the different codes/files.

Cheers,
Dave

riasatmorshed, 2024-10-20:
@dav-og, 

Sorry for replying late- somehow, I missed the notification. Thanks for taking the time to read through this long question and patiently answering them.

closed 2024-10-20 by riasatmorshed